JakobEngel/dso #4: Remove unnecessary null pointer checks

elfring, 2016-11-14:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [AccumulatedSCHessianSSE](https://github.com/JakobEngel/dso/blob/f6e43e277a8e6f606265c1d90fd7549e475e6ea6/src/OptimizationBackend/AccumulatedSCHessian.h#L54 "Destructor for AccumulatedSCHessianSSE")
- [EnergyFunctional::setAdjointsF](https://github.com/JakobEngel/dso/blob/f6e43e277a8e6f606265c1d90fd7549e475e6ea6/src/OptimizationBackend/EnergyFunctional.cpp#L44 "EnergyFunctional::setAdjointsF function")
- [PrepImageItem:release](https://github.com/JakobEngel/dso/blob/f6e43e277a8e6f606265c1d90fd7549e475e6ea6/src/util/DatasetReader.h#L92 "PrepImageItem::release function")

JakobEngel, 2016-11-15:
Huh, I didn't actually know that you can call delete on a nullpointer without blowing things up...  that does simplify things ^^.

closed 2016-11-22 by JakobEngel

elfring, 2016-11-22:
Which commits will contain [corresponding software improvements](http://en.cppreference.com/w/cpp/language/delete "Description for delete expressions")?