JakobEngel/dso #5: Improve exception safety with smart pointers

elfring, 2016-11-14:
Would you like to wrap any pointer data members with the template class "[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for usage of smart pointers")"?

Update candidates:
- [CoarseInitializer](https://github.com/JakobEngel/dso/blob/f6e43e277a8e6f606265c1d90fd7549e475e6ea6/src/FullSystem/CoarseInitializer.cpp#L44 "CoarseInitializer class")
- [ImageAndExposure](https://github.com/JakobEngel/dso/blob/f6e43e277a8e6f606265c1d90fd7549e475e6ea6/src/util/ImageAndExposure.h#L45 "ImageAndExposure class")
- [PixelSelector](https://github.com/JakobEngel/dso/blob/f6e43e277a8e6f606265c1d90fd7549e475e6ea6/src/FullSystem/PixelSelector2.cpp#L41 "PixelSelector class")

JakobEngel, 2016-11-15:
Humm, the total lack of c++11 features like smart pointers is mostly due to them, in combination with  STL containers, breaking my IDE's indexing, which I rely on frequently (or I just havn't figured out how to set up eclipse / KDevelop properly). 
Once that works more reliably, I'm happy to move to a somewhat more modern programming style ;).

closed 2016-11-22 by JakobEngel

elfring, 2016-11-22:
Would you like to reconsider the usage of current C++ functionality once more?

JakobEngel, 2016-11-22:
As soon as smart pointers & STL containers work reliably with QTCreator's indexing & code completion functionality, I'll start using them. See e.g. https://bugreports.qt.io/browse/QTCREATORBUG-17111 (and many other, related bugreports).

I do want to remove the boost dependency, and use the std mutex / thread things instead some time in the future.