JakobEngel/dso #6: dso_dataset exception on mac

hlzz, 2016-11-15:
I previously submitted a patch for dso to work on mac. The build went smoothly and I've got impressive results on a linux machine. But it turned out that dso_dataset cannot run on mac. The following log messages are given for your reference:

Tianweis-Macbook:bin STW$ ./dso_dataset files=/Users/STW/Downloads/sequence_38/images calib=/Users/STW/Downloads/sequence_38/camera.txt gamma=/Users/STW/Downloads/sequence_38/pcalib.txt vignette=/Users/STW/Downloads/sequence_38/vignette.png preset=0 mode=0
loading data from /Users/STW/Downloads/sequence_38/images!
loading calibration from /Users/STW/Downloads/sequence_38/camera.txt!
loading gammaCalib from /Users/STW/Downloads/sequence_38/pcalib.txt!
loading vignette from /Users/STW/Downloads/sequence_38/vignette.png!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
PHOTOMETRIC MODE WITH CALIBRATION!
Reading Calibration from file /Users/STW/Downloads/sequence_38/camera.txt ... found!
found ATAN camera model, building rectifier.
Input resolution: 1280 1024
In: 0.535719 0.669567 0.493249 0.500409 0.897966
Out: Crop
Output resolution: 640 480
finding CROP optimal new model!
initial range: x: -1.1372 - 1.1901; y: -0.8311 - 0.8292!
iteration 00001: range: x: -1.1315 - 1.1841; y: -0.8311 - 0.8292!
iteration 00002: range: x: -1.1258 - 1.1782; y: -0.8311 - 0.8292!
iteration 00003: range: x: -1.1258 - 1.1782; y: -0.8270 - 0.8251!
iteration 00004: range: x: -1.1258 - 1.1782; y: -0.8228 - 0.8209!
iteration 00005: range: x: -1.1258 - 1.1782; y: -0.8228 - 0.8209!

Rectified Kamera Matrix:
 277.34       0 312.234
      0 291.402 239.777
      0       0       1

Reading Photometric Calibration from file /Users/STW/Downloads/sequence_38/pcalib.txt
Reading Vignette Image from /Users/STW/Downloads/sequence_38/vignette.png
Successfully read photometric calibration!
got 3330 images and 3330 timestamps and 3330 exposures.!
ImageFolderReader: got 3330 files in /Users/STW/Downloads/sequence_38/images!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
START PANGOLIN!
2016-11-15 15:09:17.588 dso_dataset[34151:3705298] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fffc56316fb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fffd9e38a2a objc_exception_throw + 48
	2   AppKit                              0x00007fffc394450c -[NSApplication(NSEvent) shouldBeTreatedAsInkEvent:] + 0
	3   libpangolin.dylib                   0x000000010d6d737f -[PangolinNSApplication run_step] + 79
	4   libpangolin.dylib                   0x000000010d6d6d7c _ZN8pangolin9OsxWindowC2ERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEiib + 300
	5   libpangolin.dylib                   0x000000010d6d6b80 _ZN8pangolin19CreateWindowAndBindENSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEEiiRKNS_6ParamsE + 384
	6   dso_dataset                         0x000000010d55f4d5 _ZN3dso6IOWrap17PangolinDSOViewer3runEv + 133
	7   libboost_thread-mt.dylib            0x000000010d611c14 _ZN5boost12_GLOBAL__N_112thread_proxyEPv + 164
	8   libsystem_pthread.dylib             0x00007fffda92caab _pthread_body + 180
	9   libsystem_pthread.dylib             0x00007fffda92c9f7 _pthread_body + 0
	10  libsystem_pthread.dylib             0x00007fffda92c221 thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6

It seems that the problem involves pangolin. Any idea how to fix it?

JakobEngel, 2016-11-16:
Humm, yes, that seems like a Pangolin issue (or an issue with how I'm using it).

sunstarchan, 2016-11-18:
@JakobEngel I've waiting for a long time to see you publish the DSO code. Thanks a lot. 

I also run into this problem. And the root cause is that OpenCV windows for display internal result must be the main thread. As you can see the error information: reason: 'nextEventMatchingMask should only be called from the Main Thread!'

I also reported this problem to OpenCV. The comment from them is 

    Looks like you use Cocoa window backend, there could be problems when working with windows and views from different threads: [Thread Safety Summary](https://developer.apple.com/library/mac/documentation/Cocoa/Conceptual/Multithreading/ThreadSafetySummary/ThreadSafetySummary.html)

    I have doubts that thread-safe window handling can be implemented within current highgui interface using [Cocoa backend](https://github.com/Itseez/opencv/blob/master/modules/highgui/src/window_cocoa.mm). If you have different opinion - feel free to post your thoughts here.

    You can also try to use [Qt backend](https://github.com/Itseez/opencv/blob/master/modules/highgui/src/window_QT.cpp) (build OpenCV with Qt support), it should support working with windows from threads other than main.

https://github.com/opencv/opencv/issues/6039#issuecomment-181359159

And I also meet this issue when running ORB_SLAM, after I move the cv::imshow to main thread, the crash goes away. So in order to run on Mac, we should move the display into main thread.

JakobEngel, 2016-11-18:
I'm actually not using openCV for visualization (at least not per default), exactly because I kept getting issues reported due to those threading issues. the problem is likely the same though:

Apparently MacOS requires all rendering / GUI-related functions to be called not only from *the same* thread (which I believe I'm doing), but needs them to be called from *the main* thread of the process. I will have a look if there is an elegant way of solving that (essentially moving the main(...) stuff into a dedicated thread, and call PangolinDSOViewer::run() from the processes main thread) - although that is somewhat ugly.
In the meantime, a (non-ideal) workaround should be to disable the GUI (-nogui).

Also, If anyone knows of a fix to Pangolin that will allow it to open a window & render stuff from a non-main-thread (which I believe should be possible on MacOS), please let me know!

lukasvst, 2016-11-21:
I also ran into this issue a few weeks ago and moving the GUI part into the main thread solved the issue.
While it messes up the code structure a little bit it does not take long to implement it and since then DSO runs perfectly on Mac OS.
@JakobEngel I could create a pull request for this code although it is not a big change.

JakobEngel, 2016-11-22:
should be fixed. can you confirm & close the issue?
@lukasvst  @hlzz 
thanks!

hlzz, 2016-11-22:
Yes. The program can run for now. But it would still crash when any of the following three options are checked:

showFramesWindow
showFullTracking
showCoarseTracking

Here is the log message for your reference:

> Reading Photometric Calibration from file /Users/STW/Downloads/sequence_38/pcalib.txt
> Reading Vignette Image from /Users/STW/Downloads/sequence_38/vignette.png
> Successfully read photometric calibration!
> got 3330 images and 3330 timestamps and 3330 exposures.!
> ImageFolderReader: got 3330 files in /Users/STW/Downloads/sequence_38/images!
> using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
> START PANGOLIN!
> Initialization: keep 21.7% (need 2000, have 9205)!
> INITIALIZE FROM INITIALIZER (1995 pts)!
> SPARSITY:  MinActDist 1.900000 (need 2000 points, have 1995 points)!
> OPTIMIZE 1995 pts, 1995 active res, 0 lin res!
> Initial Error       	A(271220.825474)=(AV inf). Num: A(0) + M(0); ab 0.000000 0.000000!
> STEPS: A 0.0; B 4.8; R 3.7; T 19.4. 	ACCEPT 0 (L -1.00, dir nan, ss 1.0): 	A(220350.908493)=(AV 4.517). Num: A(1350) + M(0); ab -0.000023 -0.408337!
> STEPS: A 0.0; B 0.8; R 5.5; T 16.6. 	ACCEPT 1 (L -1.60, dir -0.10, ss 1.0): A(193502.530315)=(AV 4.124). Num: A(1422) + M(0); ab -0.000026 -0.474465!
> STEPS: A 0.0; B 0.3; R 4.9; T 13.2. 	ACCEPT 2 (L -2.20, dir 0.65, ss 1.0): 	A(190411.950635)=(AV 4.098). Num: A(1417) + M(0); ab -0.000026 -0.495699!
> STEPS: A 0.0; B 0.1; R 3.9; T 9.8. 	ACCEPT 3 (L -2.81, dir 0.98, ss 1.0): 	A(188954.762744)=(AV 4.086). Num: A(1415) + M(0); ab -0.000027 -0.501420!
> STEPS: A 0.0; B 0.1; R 2.7; T 6.4. 	ACCEPT 4 (L -3.41, dir 1.00, ss 1.0): 	A(188508.451043)=(AV 4.079). Num: A(1416) + M(0); ab -0.000027 -0.508361!
> STEPS: A 0.0; B 0.0; R 2.1; T 4.6. 	ACCEPT 5 (L -4.01, dir 0.98, ss 1.0): 	A(188267.492799)=(AV 4.075). Num: A(1417) + M(0); ab -0.000027 -0.508090!
> STEPS: A 0.0; B 0.0; R 1.5; T 3.1. 	ACCEPT 6 (L -4.61, dir 1.00, ss 1.0): 	A(188231.477409)=(AV 4.073). Num: A(1418) + M(0); ab -0.000027 -0.510085!
> STEPS: A 0.0; B 0.0; R 1.1; T 2.2. 	REJECT 7 (L -5.21, dir 0.98, ss 1.0): 	A(188270.271836)=(AV 4.074). Num: A(1418) + M(0); ab -0.000027 -0.508798!
> STEPS: A 0.0; B 0.0; R 0.8; T 1.4. 	REJECT 8 (L -5.82, dir 0.99, ss 1.0): 	A(188765.669262)=(AV 4.079). Num: A(1418) + M(0); ab -0.000027 -0.511647!
> STEPS: A 0.0; B 0.1; R 1.4; T 3.6. 	ACCEPT 9 (L -6.42, dir 0.99, ss 1.0): 	A(188246.814878)=(AV 4.076). Num: A(1416) + M(0); ab -0.000027 -0.505482!
> STEPS: A 0.0; B 0.0; R 2.6; T 6.7. 	ACCEPT 10 (L -7.02, dir 1.00, ss 1.0): A(188207.493248)=(AV 4.076). Num: A(1416) + M(0); ab -0.000027 -0.504278!
> STEPS: A 0.0; B 0.0; R 1.6; T 3.9. 	REJECT 11 (L -7.62, dir 0.99, ss 1.0): A(188243.277633)=(AV 4.076). Num: A(1416) + M(0); ab -0.000027 -0.502517!
> STEPS: A 0.0; B 0.0; R 1.2; T 2.7. 	REJECT 12 (L -8.22, dir 1.00, ss 1.0): A(188780.399215)=(AV 4.082). Num: A(1416) + M(0); ab -0.000027 -0.506729!
> STEPS: A 0.0; B 0.0; R 1.0; T 2.2. 	REJECT 13 (L -8.83, dir 0.99, ss 1.0): A(188911.018883)=(AV 4.085). Num: A(1415) + M(0); ab -0.000027 -0.505038!
> STEPS: A 0.0; B 0.0; R 0.7; T 1.4. 	REJECT 14 (L -9.43, dir 0.99, ss 1.0): A(189023.176217)=(AV 4.086). Num: A(1415) + M(0); ab -0.000027 -0.508112!
> LOG 9: 4.086 fine. Res: 1415 A, 0 L, 0 M; (0 / 0) forceDrop. a=-0.000027, b=-0.508112. Window 9 (2)
> Coarse Tracker tracked ab = -0.014762 0.641270 (exp 10.081544). Res 2.825979!
> Coarse Tracker tracked ab = -0.010591 0.181534 (exp 10.081544). Res 3.435972!
> Coarse Tracker tracked ab = -0.010527 0.026119 (exp 10.081544). Res 4.082731!
> Coarse Tracker tracked ab = -0.007131 -0.388757 (exp 10.081544). Res 4.996680!
> Coarse Tracker tracked ab = -0.013262 0.021158 (exp 10.081544). Res 5.626656!
> Coarse Tracker tracked ab = -0.007376 -0.659720 (exp 10.081544). Res 5.857679!
> SPARSITY:  MinActDist 1.400000 (need 2000 points, have 1415 points)!
> OPTIMIZE 2013 pts, 4019 active res, 0 lin res!
> Initial Error       	A(476832.231487)=(AV 6.490). Num: A(1415) + M(0); ab -0.007376 -0.659720!
> STEPS: A 0.7; B 2.2; R 73.5; T 178.9. 	ACCEPT 0 (L -1.00, dir nan, ss 1.0): 	A(401941.328279)=(AV 3.672). Num: A(3727) + M(0); ab -0.000018 -0.863199!
> STEPS: A 0.0; B 0.2; R 49.1; T 109.2. 	ACCEPT 1 (L -1.60, dir 0.98, ss 1.0): 	A(362523.389061)=(AV 3.456). Num: A(3794) + M(0); ab -0.000021 -0.875217!
> STEPS: A 0.0; B 0.2; R 23.0; T 46.9. 	ACCEPT 2 (L -2.20, dir 0.65, ss 1.0): 	A(343117.786523)=(AV 3.342). Num: A(3840) + M(0); ab -0.000022 -0.859619!
> STEPS: A 0.0; B 0.1; R 15.1; T 30.1. 	ACCEPT 3 (L -2.81, dir 0.98, ss 1.0): 	A(329160.044208)=(AV 3.264). Num: A(3862) + M(0); ab -0.000023 -0.860462!
> STEPS: A 0.0; B 0.0; R 7.9; T 16.9. 	ACCEPT 4 (L -3.41, dir 0.95, ss 1.0): 	A(325341.854631)=(AV 3.242). Num: A(3869) + M(0); ab -0.000023 -0.857959!
> STEPS: A 0.0; B 0.2; R 4.4; T 7.3. 	ACCEPT 5 (L -4.01, dir 0.95, ss 1.0): 	A(323834.383859)=(AV 3.234). Num: A(3870) + M(0); ab -0.000024 -0.876311!
> STEPS: A 0.0; B 0.0; R 2.0; T 5.6. 	REJECT 6 (L -4.61, dir 0.70, ss 1.0): 	A(326344.160153)=(AV 3.247). Num: A(3870) + M(0); ab -0.000023 -0.876753!
> STEPS: A 0.0; B 0.1; R 0.8; T 1.2. 	REJECT 7 (L -5.21, dir -0.56, ss 1.0): A(326544.014679)=(AV 3.248). Num: A(3868) + M(0); ab -0.000024 -0.891243!
> STEPS: A 0.0; B 0.1; R 0.9; T 2.5. 	REJECT 8 (L -5.82, dir -0.52, ss 1.0): A(328701.510005)=(AV 3.260). Num: A(3865) + M(0); ab -0.000024 -0.885613!
[... cut ...]
> 	8   dso_dataset                         0x00000001066eddc7 _ZN3dso10FullSystem14addActiveFrameEPNS_16ImageAndExposureEi + 1623
> 	9   dso_dataset                         0x00000001066e0f42 _ZNSt3__114__thread_proxyINS_5tupleIJZ4mainE3$_0EEEEEPvS4_ + 1282
> 	10  libsystem_pthread.dylib             0x00007fffda92caab _pthread_body + 180
> 	11  libsystem_pthread.dylib             0x00007fffda92c9f7 _pthread_body + 0
> 	12  libsystem_pthread.dylib             0x00007fffda92c221 thread_start + 13
> )
> libc++abi.dylib: terminating with uncaught exception of type NSException
> Abort trap: 6
>

JakobEngel, 2016-11-23:
Great! Yes, that is somewhat expected, since those options create additional, openCV-based display windows and really are only left-overs from when I was using OpenCV for visualization.
It's unlikely I'll fix that - just don't use the buttons, they are only for debugging anyway ;).

If I find the time, I may integrate that functionality (which essentially visualize the residual image for initial direct image alignment, the color-coded points across their respective keypoints, and all projected points across all keyframes in many windows) into the Pangolin based viewer - which will fix this issue.

closed 2016-11-23 by JakobEngel