JakobEngel/dso #12: Test dso on KITTI Dataset 00

TimingSpace, 2016-11-19:
I want try to use dso on KITTI dataset, and I have tested it on KITTI sequence 00
set camera.txt as
0.5793	1.9119	0.4893 0.4926 0
none
1241 376

then run dso by ./bin/dso_dataset files=$path/image_0 calib=$path/camera.txt  preset=0 mode=2
and do not set pcalib and vignette.

and got an error

/home/shikw/Documents/Dataset/00_gray/
loading data from /home/shikw/Documents/Dataset/00_gray//image_0!
loading calibration from /home/shikw/Documents/Dataset/00_gray//camera.txt!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
PHOTOMETRIC MODE WITH PERFECT IMAGES!
Reading Calibration from file /home/shikw/Documents/Dataset/00_gray//camera.txt ... found!
found OpenCV camera model, building rectifier.
Input resolution: 1241 376
In: 0.579300 1.911900 0.489300 0.492600 0.000000 0.000000 0.000000 0.000000
Out: Full
Output resolution: 1241 376

Rectified Kamera Matrix:
718.911       0 606.721
      0 718.874 184.718
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file 
PhotometricUndistorter: Could not open file!
set timestamps and exposures to zero!
set EXPOSURES to zero!
got 403 images and 0 timestamps and 0 exposures.!
ImageFolderReader: got 403 files in /home/shikw/Documents/Dataset/00_gray//image_0!
using pyramid levels 0 to 0. coarsest resolution: 1241 x 376!

===============WARNING!===================
 using not enough pyramid levels.
Consider scaling to a resolution that is a multiple of a power of 2.

===============WARNING!===================
 I need higher resolution.
I will probably segfault.
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

Deejep, 2016-11-19:
There is a calib file coming along with the images. Here is what's inside:

```
P0: 7.188560000000e+02 0.000000000000e+00 6.071928000000e+02 0.000000000000e+00 0.000000000000e+00 7.188560000000e+02 1.852157000000e+02 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00 1.000000000000e+00 0.000000000000e+00
P1: 7.188560000000e+02 0.000000000000e+00 6.071928000000e+02 -3.861448000000e+02 0.000000000000e+00 7.188560000000e+02 1.852157000000e+02 0.000000000000e+00 0.000000000000e+00 0.000000000000e+00 1.000000000000e+00 0.000000000000e+00
P2: 7.188560000000e+02 0.000000000000e+00 6.071928000000e+02 4.538225000000e+01 0.000000000000e+00 7.188560000000e+02 1.852157000000e+02 -1.130887000000e-01 0.000000000000e+00 0.000000000000e+00 1.000000000000e+00 3.779761000000e-03
P3: 7.188560000000e+02 0.000000000000e+00 6.071928000000e+02 -3.372877000000e+02 0.000000000000e+00 7.188560000000e+02 1.852157000000e+02 2.369057000000e+00 0.000000000000e+00 0.000000000000e+00 1.000000000000e+00 4.915215000000e-03
Tr: 4.276802385584e-04 -9.999672484946e-01 -8.084491683471e-03 -1.198459927713e-02 -7.210626507497e-03 8.081198471645e-03 -9.999413164504e-01 -5.403984729748e-02 9.999738645903e-01 4.859485810390e-04 -7.206933692422e-03 -2.921968648686e-01
```

If you figure out how to convert it to one of the DSO compatible calib inputs, please tell us.

Cheers!

TimingSpace, 2016-11-21:
@Deejep when I use 
Input resolution: 1241 376
In: 718.856 718.856 607.1928 185.2157 0.000000 0.000000 0.000000 0.000000
Out: crop
Output resolution: 1241 376

and preset = 3 mode =2 

svo can run without error

JakobEngel, 2016-11-22:
As TimingSpace mentioned, Try something like
-------------------------
718.856 718.856 607.1928 185.2157 0.000000
1241 376
crop
1232 376
----------------------------

That should allow DSO to rescale / crop the image to have appropriate resolution ("none" means there is no rectification, and thereby also no re-scaling / cropping is performed, thus the error message).

To explain a bit further:
DSO requires the image width / height to satisfy
width = 2^n * ws
height = 2^n * hs
with n some integer, and ws, hs <= 100 pixel, to allow easy computation of the scale-pyramid. 
There is no real reason for this (You could just omit the last row / column when down-scaling, thereby allowing uneven sizes). The only reason this is not implemented is because all standard image sizes satisfy this constraint, and it makes the code more bloated (and I'm lazy).

"-mode=2/3" simply overwrites the last line of the calibration file with "424 320".

closed 2016-11-29 by JakobEngel

ashwinjoseph95, 2024-09-23:
THis issue can be solved by jsut resizing the image to image size divisible by 2. Thw warning mentions it: _Consider scaling to a resolution that is a multiple of a power of 2._
I tired reszing image to 1280x384 and my new calibration file is as follows:
Pinhole 741.428 734.146 626.258 189.1552 0
1280 384
none
1280 384
it seems to work! though performance of the algorithm with dataset kitti 00 does not seem to be the best