JakobEngel/dso #13: Hint on how to add support for rolling shutter cameras?

lexvandersluijs, 2016-11-19:
Hi,

Thanks for publishing this amazing work, my team and I are looking forward to running it in real time ourselves!

It's clear that the algorithm is best suited for global shutter cameras. At the same time, in current mobile devices these are hard to find.  

Theoretically it should be possible to add support for rolling shutter cameras, since the optimization is done on both the photometric and geometric distortion, and the geometric distortion caused by a rolling shutter is well known for a projection of a 3D point when 
- the 3D point is known (well enough)
- the sensor readout time is known (usually a simple function of the frame rate)
- the readout-direction of the sensor is known (bottom-to-top or top-to-bottom, in my experience usually it's the first)
- the camera's pose is known
- the camera's rotational and positional velocity are known

(where the term 'is known' can be replaced by 'is being estimated', depending on the context)

Probably the first three are already present somewhere in the optimization. Can you estimate if it is feasible to add the velocities as well? Then the location of the projection of the 3D point, depending on its Y-location on the sensor can be estimated completely (or conversely: the camera's rotational and positional velocities can be estimated, thereby accounting for the rolling shutter distortion).

 That would make the system even more revolutionary than it already is!

BR,
Lex

JakobEngel, 2016-11-22:
Hi Lex,

First, you probably also will want to add an IMU at the same time, to better capture high-frequent motion (and since it is present in any spartphone).

That said, I believe a more promising approach than assuming linear motion (constant rotational / positional velocity) during the readout timespan is a Spline-based approach (see Spline Fusion http://stevenlovegrove.com/?id=46, and also https://vision.in.tum.de/~kerl/kerl_etal_iccv2015_webpage/), since it allows to approximate arbitrary continuous motion over time.

In practice, both approaches will increase the required compute, since they likely increase the number of pose parameters optimized over, which has quadratic impact on compute - also, now each row (or approximated, block of rows) will have it's own pose, making some compute optimizations employed  impossible / less effective (e.g., (1) pre-computing rotation matrices for fast point warping, or pulling d[T_hostToTarget]/d[T_hostToWorld, T_targetToWorld] out of the sums, and (2) the inverse depth initialization, where exploit fact that the epipolar lines are straight to avoid having to project over and over again).

In summary, I believe there is nothing prohibiting incorporation of a rolling shutter, other than it making the code much more complex. At the same time, compute will suffer, and / or you have to employ aggressive approximations (e.g., grouping rows), to preserve real-time capability. Note that all such approximations will trade off additional geometric error for faster compute, which can have a severe impact on accuracy for direct approaches.

Best,
Jakob

lexvandersluijs, 2016-11-22:
Hi Jakob,

Thanks a lot for your thoughtful and detailed reply, and for the Spline Fusion suggestion.
We are interested in the system for handheld AR, so a relatively high framerate is required. This may be difficult with the Spline Fusion approach?

I was indeed expecting that support for a rolling shutter would be tremendously difficult to implement, given that with a rolling shutter the projection of a point also depends on its (resulting) y-coordinate and the camera's velocities. This would give rise to some 'interesting' Jacobians and/or Hessians.

In handheld or head-worn tracking the largest geometric distortions are caused by rotations (one does not typically lunge forward or left/right while holding a tablet for example), so we could try unwarping the image by using the gyroscope readings. It should be interesting to see how the system behaves in that case.

BR,
Lex

JakobEngel, 2016-11-24:
Yes, there is always the option of compensating for the rotational component of the distortion via a pre-computation step - however, again that is an approximation which will will leave you with significant geometric distortions, which is generally bad.... as shown experimentally in the paper, indirect methods tend to be more robust to such geometric distortions. However, I cannot say how severe the effect will be.

Regarding the Spline Fusion: I believe the framerate of tracking is not in fact the critical point, rather it is the delay of the pose estimate that is critical -- which you typically will want to compensate for by predicting ahead in time, using some motion model. The Spline representation may offer you a better way of doing this prediction (assuming over maybe 50-100ms, human motion can well be approximated by a spline, which I don't know).

Cheers, Jakob

closed 2016-12-03 by JakobEngel