JakobEngel/dso #14: Error  9 reading archive sequence_01/images.zip!

chiragmajithia, 2016-11-19:
Hello, I built the dso package successfully and downloaded sequence_01 from tum's mono_dataset.. However when I run dso_dataset, it gives me Error 9 reading archive sequence_01/images.zip. I have extracted the sequence in dso/build/bin/ folder.

Please let me know if I am doing something wrong.

Regards,
Chirag 

`bin/dso_dataset files=sequence_01/images.zip calib=sequence_01/camera.txt gamma=sequence_01/pcalib.txt vignette=sequence_01/vignette.png preset=0 mode=0
loading data from sequence_01/images.zip!
loading calibration from sequence_01/camera.txt!
loading gammaCalib from sequence_01/pcalib.txt!
loading vignette from sequence_01/vignette.png!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
PHOTOMETRIC MODE WITH CALIBRATION!
ERROR 9 reading archive sequence_01/images.zip!
`

chiragmajithia, 2016-11-19:
I ran it with mode = 1, it worked!  and then I ran it with mode = 0 and it worked again!

closed 2016-11-19 by chiragmajithia

NikolausDemmel, 2017-12-01:
Not sure what this error 9 is exactly, but two things to try:

- Run it with extracted images.
- If you are on Ubuntu 14.04., you need a more recent libzip as per the readme. Maybe make sure to uninstall and libzip package from apt, and then follow the steps from the readme again to compile and install libzip.

nunuwin, 2017-12-01:
my Ubuntu is 16.04 and my error is as follows:

nu@nu-VirtualBox:~/dso/build$ bin/dso_dataset \
> files=home/sequence_01/images.zip \
> calib=home/sequence_01/camera.txt \
> gamma=home/sequence_01/pcalib.txt \
> vignette=home/sequence_01/vignette.png \
> preset=0 \
> mode=1
loading data from home/sequence_01/images.zip!
loading calibration from home/sequence_01/camera.txt!
loading gammaCalib from home/sequence_01/pcalib.txt!
loading vignette from home/sequence_01/vignette.png!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
PHOTOMETRIC MODE WITHOUT CALIBRATION!
ERROR 9 reading archive home/sequence_01/images.zip!
nu@nu-VirtualBox:~/dso/build$ 
.......
so, is it same?

nunuwin, 2017-12-01:
Thanks Nikolaus, and I would like to know, when I installed dso I make this 'sudo apt-get install zlib1g-dev' do you mean this one? And now I check libzip package in my ubuntu and installed is but it is like that

'nu@nu-VirtualBox:~$ dpkg -L libzip-dev
dpkg-query: package 'libzip-dev' is not installed
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
nu@nu-VirtualBox:~$ sudo apt-get install libzip-dev
E: Could not get lock /var/lib/dpkg/lock - open (11: Resource temporarily unavailable)
E: Unable to lock the administration directory (/var/lib/dpkg/), is another process using it?

Am I need to uninstall zlib1g-dev?

NikolausDemmel, 2017-12-01:
Did you manage to run with an unzipped folder of images?

`zlib1g-dev` from apt you need in any case, that is fine.

You should be able to build DSO with `libzip-dev` from apt on 16.04., but if you have compiled and installed it manually as well from the sources in the DSO repository (as described in the README), you might have a conflict there.

I would either uninstall `libzip-dev` from apt and install it again from own compilation, as described in the readme, or the other way round.

NikolausDemmel, 2017-12-01:
It seems ERROR 9 also indicates if the file is not found. You might be missing a leading slash in your paths, or are they relative? Make sure the file is found on the path you pass to dso.

nunuwin, 2017-12-04:
yes, I managed to run with an unzipped folder of images, and it is ok.
And another one is camera.txt cant find he said.
But I checked camera.txt, it is there.
So that means... for which?

NikolausDemmel, 2017-12-04:
As mentioned, I guess the path is missing a slash at the beginning. If the file is not found, it is likely because the path is wrong, (or maybe because of read permissions).

nunuwin, 2017-12-13:
I checked, but slash is not missed. So, how to find correct path? Please..

nunuwin, 2017-12-13:
I run these commands after installed dso directly, is is need to download or not dso_dataset from 'https://vision.in.tum.de/mono-dataset'

NikolausDemmel, 2017-12-13:
If you want to run it on a dataset, yes, you need to download it.

The arguments files, calib, gamma, vignette are paths to files / folders on your system.

NikolausDemmel, 2017-12-13:
What does the following command output (run from the `~/dso/build` folder):

```
ls home/sequence_01/images.extracted
```

nunuwin, 2017-12-27:
If I run this command 'ls home/sequence_01/images.extracted '
reply is like that
nu@nu-VirtualBox:~/dso/build$ ls home/sequence_01/images.extracted
ls: cannot access 'home/sequence_01/images.extracted': No such file or directory

nunuwin, 2017-12-27:
Sorry I can find it

NikolausDemmel, 2017-12-27:
> If I run this command 'ls home/sequence_01/images.extracted '
reply is like that
>```
>nu@nu-VirtualBox:~/dso/build$ ls home/sequence_01/images.extracted
>ls: cannot access 'home/sequence_01/images.extracted': No such file or directory
>```

What you see is not surprising. You need to make sure the paths you provide are absolute or relative paths to the according files or folders on your system.

> Sorry I can find it

So you resolved the issue?

yolingen23, 2019-06-19:
This error occurs when I try to run the program.

`gurturk@gurturk:~/dso$ bin/dso_dataset \
> files =/home/gurturk/D
DBoW2-master/ Desktop/      Documents/    Downloads/    
> files =/home/gurturk/Downloads/sequence_17
sequence_17/     sequence_17.zip  
> files =/home/gurturk/Downloads/sequence_17/images.zip \
> calib=/home/gurturk/Downloads/sequence_17/camera.txt \
> gamma=/home/gurturk/Downloads/sequence_17/pcalib.txt \
> vignette=/home/gurturk/Downloads/sequence_17/vignette.png \
> preset=0\
> mode=0
bash: bin/dso_dataset: No such file or directory
`

how can i run dso?

Virat-Akhil, 2020-06-16:
I got similar error.  I have extracted images folder, and also the sequence_01 file is in same location, but still it says this error. Could someone help me
```
bin/dso_dataset \files=sequence_01.zip/sequence_01/images.zip \calib=sequence_01.zip/sequence_01/camera.txt \gamma=sequence_01.zip/sequence_01/pcalib.txt \vignette=sequence_01.zip/sequence_01/vignette.png \preset=0 \mode=1
loading data from sequence_01.zip/sequence_01/images.zip!
loading calibration from sequence_01.zip/sequence_01/camera.txt!
loading gammaCalib from sequence_01.zip/sequence_01/pcalib.txt!
loading vignette from sequence_01.zip/sequence_01/vignette.png!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
PHOTOMETRIC MODE WITHOUT CALIBRATION!
ERROR 5 reading archive sequence_01.zip/sequence_01/images.zip!
```

Virat-Akhil, 2020-06-16:
> This error occurs when I try to run the program.
>
> `gurturk@gurturk:~/dso$ bin/dso_dataset \
>
>
> how can i run dso?

Try to give correct path for DSO to run.