JakobEngel/dso #15: run dso with myself dataset, read camera.txt error

dq5070410, 2016-11-21:
hi,
I want to run the dos with myself dataset, the dataset conclude a folder of images and a camera.txt,
it format is
763.250	765.168	310.936	230.990 0.09475 0.02371 -0.002102 0.001759
640 480 crop 640 480
then execute the command 
 ./dso_dataset files=/mypath/DSO/bagfile/img/ calib=/mypath/DSO/bagfile/camera.txt mode=1

but it appearance a error like this
loading data from /mypath/DSO/bagfile/img/!
loading calibration from /mypath/DSO/bagfile/camera.txt!
PHOTOMETRIC MODE WITHOUT CALIBRATION!
Reading Calibration from file /mypath/DSO/bagfile/camera.txt ... found!
found OpenCV camera model, building rectifier.
Input resolution: 640 480
In: 763.250000 765.168000 310.936000 230.990000 0.094750 0.023710 -0.002102 0.001759
Out: Failed to Read Output pars... not rectifying.
Segmentation fault (core dumped)

anyone can give some helps?

cheers

closed 2016-11-21 by dq5070410

DxsGeorge, 2017-07-27:
@dq5070410 Hi! What was the problem?

mrwtakev, 2018-03-16:
@dq5070410 你好，我也遇到了这个问题，请问你是怎么解决的。谢谢。

dq5070410, 2018-03-16:
@mrwtakev it's a long time and I almost forgot this problem.
 It should be the images name format error.

mrwtakev, 2018-03-19:
@dq5070410 Thank you. Do you run your own dataset successfully? I'm running my own dataset, the effect of reconstruction is poor and it is completely invisible. I suspect there are problems with the parameters in my camera.txt. Can you give me some advice?Thank you.

dq5070410, 2018-03-20:
@mrwtakev  I run my own dataset success. DSO require a global shutter camera，and exposure not change. And you should move slowly and texture rich but not duplicate texture