JakobEngel/dso #16: Inv. response function U is not monotonic

chaizheng2157, 2016-11-22:
Hello there, thanks for open-source.
I have a question that the inverse response function U is not monotonic based on my dataset (more than 1000 images, static scene and different exposure time from 0.05ms to 20ms). You said in this case, "it" should be smoothed or perturbed on your paper, and what is "it"?
I have modified some code like "G[255]" to "Gmax"(which is the max value in the G[]), is there some other codes I need to modified?
I'm sure that there are many different objects in the static scene, and the exposure time are incremental.  Are there some processing like adding blur or noise while capturing the images?  Look forward to your response and help. Thank you.

Best wishes.

JakobEngel, 2016-11-22:
humm, that is interesting - how does your inverse response function look like (i.e., the content of [....]/photoCalibResult/pcalib.txt)? You should not have to modify any code; most likely your
inverse response function is just a little non-monotonic somewhere, which is easy to fix.

"it" refers to exactly the (discretized) function in that file, it should contain 256 *increasing* values. The reason why this is required is, that I'm doing the point-selection on the original image (i.e., in pixel intensity space), and since only the photometrically "corrected" images are passed into DSO, I need to undo the correction (see HessianBlocks.cpp#L184). For that I need to invert the inverse response. A bit complicated and not very nice from a software engineering perspective, but let's just say it's grown historically ;).

You can also use the response calibration approach by Debevec & Malik ( http://www.pauldebevec.com/Research/HDR/debevec-siggraph97.pdf ) which includes a regularizer to enforce smoothness, I believe they have an open-source implementation somewhere.

chaizheng2157, 2016-11-23:
Thanks, I have found the problem, that is the error of reading the  images. I made a mistake that saved the images as "1.png", "2.png", not as "00001.png", "00002.png". The order of reading the images is wrong so that they do not match with the corresponding exposure times. 
Anyway, thank you very much. :)

JakobEngel, 2016-11-23:
:D yeah I know that problem well ^^.

closed 2016-11-23 by JakobEngel