JakobEngel/dso #18: Project exits automatically and can not save the result

zhouyanxing, 2016-11-30:
I have learned the DSO based ROS to read the camera, and transform the the project into NO_ROS version.It can be used normally, but will automatically exit encountered the following conditions :
(1)The camera has a violent shake.In fact, sometimes a slight shake also makes it exit;
(2)Running time is little longer.
(3)Light intensity change suddenly;
And I try to use "while(cvWaitKey(33) != 's')" to close the main loop and save the result, but this code don't work. I think the reason is that this GUI window isn't created by OpenCV.
So I want to know how to improve the project's stability and how to save the result.
Thanks a lot!

zhouyanxing, 2016-11-30:
I have solved the second problem, thanks...

JakobEngel, 2016-12-02:
I've made a small change such that DSO will simply say it got lost and stop tracking, instead of crashing. 

Since DSO is operating in a strict VO framework however, it will get lost as soon as a couple (or sometimes even one) frame cannot be tracked - causing it to get lost when you switch on the lights, or on violent shakes. 
The proper way around this would be to build a full SLAM system (which can be done in the same sparse & direct framework), and include proper re-localization.

closed 2016-12-03 by JakobEngel