JakobEngel/dso #19: how to save the pointcloud

fyj3266098, 2016-11-30:
if you do not have realize the function of saving pointcloud ,can you tell me which variables have save the pointcloud.

JakobEngel, 2016-12-02:
please have a look at src/IOWrapper/Output3DWrapper.h, in that file there is a short description of the classes / structs used. You can then create an Output3DWrapper implementation and incrementally create the pointcloud by hooking into the Output3DWrapper::publishKeyframes() function (again, see Output3DWrapper.h).

By default, the core DSO algorithm does not save marginalized keyframes internally to avoid ever-growing memory requirements - they are only stored for visualization in the PangolinDSOViewer.

closed 2016-12-03 by JakobEngel

fyj3266098, 2016-12-06:
I have save the point cloud in refreshPC of KeyFrameDisplay.cpp where i write the tmpVertexBuffer[vertexBufferNumPoints] to files every time and corresponding R&T which could convert the camera coordinate to world coordinate. Do you think is it right?

shikailei, 2017-12-25:
hello,can you save pointcloud now?

GentlemanCheng, 2019-04-04:
hello,can you save the dso pointcloud now?