JakobEngel/dso #20: bebop camera calibration

chiragmajithia, 2016-12-01:
Hey,
I got dso_ros running with parrot bebop - however, the dso slam is not able to extract translation - I can provide the screencast video if required. I think I am not updating the calibration parameters correctly. Could you please guide me through how to generate the calibration file from bebops default calibration?
Regards,
Chirag.
[bebop_camera_calib.txt](https://github.com/JakobEngel/dso/files/624651/bebop_camera_calib.txt)

fairf4x, 2016-12-11:
Hallo,
I have just tested DSO on my bebop with config file cam_bebop.txt.
I am running [bebop_driver](http://bebop-autonomy.readthedocs.io/en/latest/index.html) on ROS kinetic.
DSO needs to be restarted after bebop_driver node is initialized and camera starts to publish images.

I have created the calib file `cam_bebop.txt` "by hand" from `cam_bebop_yaml.txt`.
The trick is to realize (I needed to google for a while) that the `camera_matrix` in the yaml file is in fact:

fx 0  cx
0  fy cy
0  0  1

Where values fx, fy, cx,cy corresponds to the parameters of camera model in DSO calibration file.

However I still do not understand the process of creating my own photometric calibration files `vignette.png` and `pcalib.txt`. The SLAM itself seems to be working without these files but I think they might be quite important because they describe some important properties of the camera.

[cam_bebop.txt](https://github.com/JakobEngel/dso/files/644862/cam_bebop.txt)
[cam_bebop_yaml.txt](https://github.com/JakobEngel/dso/files/644863/cam_bebop_yaml.txt)

NikolausDemmel, 2017-12-07:
The questions seems mostly answered. For future reference, response and vignette calibration can be done with https://github.com/tum-vision/mono_dataset_code.

Feel free to comment and we can reopen if needed.

closed 2017-12-07 by NikolausDemmel

nunuwin, 2018-01-09:
I want to run dso with gopro video files, so I want to know can I run dso with video file and which are needed to run dso with gopro video file. This is the first experiment in image field, I can run dso with data_set. but I don't know how to run real_time and whith video. In data_set, camera.txt, pcalib.txt, groundtruth.txt, and vignette.png . If I run in real_time or my data, how to write or get these files?