JakobEngel/dso #22: Eigen Assertion Error in Initialization

airRobotFun, 2016-12-02:
@JakobEngel, any suggestions for the following error? Much Appreciated. 

[image]

JakobEngel, 2016-12-02:
Thanks! It's an Eigen alignment issue - can you try the most recent commit on the <devel> branch? It should be fixed there.

I believe it will appear depending on how the compiler optimizes the code and / or some system properties (such as 32 vs 64 bit) -- for some reason it doesn't occur on any of my systems, so I got lazy with including all the required alignment precautions. Let me know if it is fixed!

airRobotFun, 2016-12-02:
Thanks for getting after it so quick!!! 
It made it past the initialization and Pangolin opened up for a couple frame, but it crashed shortly thereafter. 

[image]

I can dig a little deeper as I did above if you'd like. let me know. 

A couple install notes for anyone else who might stumble across the same problems I had. In order to get dso to compile on my machine, I had to add the line set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -msse4.1") to the CMakeFile just after setting the SSE_FLAGS. I also had to add -lpthread to the target_link_libraries. 

I've got a 32-bit machine so after some googling when I found that someone else solved a similar problem by reinstalling a 64-bit version of Ubunutu, I gave up on the linux install and got it running on a MacBook Pro (Yosemite). Just had to replace boost_thread in target_link_libraries with explicit links to the /usr/local/lib/libboost_thread-mt.dylib and other boost dylib files because my installation (from brew install boost) wasn't supported by the mac architecture (wanted i386 instead of x64).

Also, I'm getting warnings when running make -j4 on both machines, but haven't had time to look into solving them yet since it still compiles. And also, make -j freezes the terminal.

JakobEngel, 2016-12-03:
Yes, the 32-bit machine is likely the reason. However, if implemented properly, it should work in any case, I just dont have the means of testing that, so there are probably some alignment issues left in the code.
If you can get me a stacktrace, I can probably fix that issue as well... the question is just how many more there will be ;).

Thx for the installation notes, I'll integrate it in the readme.

rongbohou, 2016-12-19:
I used the devel branch and modified some alignment issues. But it still  crashed after the initialization.
I get:
[image]
any suggestions for the above error? Thanks!!

JakobEngel, 2016-12-21:
Hey, I have added another small fix at around that location - if that doesn't do the trick, I am not sure what alignment issues may be left ^^.

mah92, 2016-12-21:
Hi, same problem on my system that you solved it's compiler error, (32bit, ububtu16.04), I will try to test your improvement as soon as possible...

mah92, 2016-12-22:
I tested your newest devel version on my 32 bit system, compiled with your latest Cmake in master code, it successfully worked for some frames but after printing:
STEPS: A 0.0; B 4.2; R 7.4; T 35.2. 	REJECT 0 (L -1.00, dir nan, ss 1.0): 	A(715867.549185)=(AV 8.730). Num: A(1174) + M(0); ab -0.000025 0.354148!

it gave me a segmentation fault:
Segmentation fault (core dumped)

Is it when changing keyframes?

mah92, 2016-12-22:
The problem is exactly occurred at (IndexThreadReduce.h, function: reduce, line: 96):
 this->callPerIndex = callPerIndex;
this function was called by (EnergyFunctional.cpp, function: EnergyFunctional::resubstituteF_MT, line: 283)

mah92, 2016-12-22:
function reduce is called successfully from many locations in code until it is called by resubstituteF_MT

mah92, 2016-12-22:
here is the trace:
#0  0xb7fdbd4c in __kernel_vsyscall ()
#1  0xb7780607 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#2  0xb7783a33 in __GI_abort () at abort.c:89
#3  0xb7779757 in __assert_fail_base (
    fmt=0xb78b7014 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x8132bd8 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAsse"..., 
    file=file@entry=0x8132ba4 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", 
    line=line@entry=78, 
    function=function@entry=0x81538a0 <_ZZN5Eigen8internal11plain_arrayIfLi4ELi0ELi16EEC1EvE19__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = float; int Size = 4; int MatrixOrArrayOptions = 0]")
    at assert.c:92
#4  0xb7779807 in __GI___assert_fail (
    assertion=0x8132bd8 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAsse"..., 
    file=0x8132ba4 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=78, 
    function=0x81538a0 <_ZZN5Eigen8internal11plain_arrayIfLi4ELi0ELi16EEC1EvE19__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = float; int Size = 4; int MatrixOrArrayOptions = 0]")
    at assert.c:101
#5  0x080d7e32 in Eigen::internal::plain_array<float, 4, 0, 16>::plain_array (
    this=0xa26b3398) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:78
#6  0x080f8717 in DenseStorage (this=0xa26b3398)
    at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:120
#7  PlainObjectBase (nbCols=1, nbRows=4, a_size=4, this=0xa26b3398)
    at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:438
#8  Matrix (other=..., this=0xa26b3398)
    at /usr/include/eigen3/Eigen/src/Core/Matrix.h:285
#9  value (this=0xa26b3398) at /usr/include/boost/bind/bind.hpp:112
#10 storage2 (this=0xa26b3388) at /usr/include/boost/bind/storage.hpp:76
#11 storage3 (this=0xa26b3388) at /usr/include/boost/bind/storage.hpp:125
#12 storage4 (this=0xa26b3388) at /usr/include/boost/bind/storage.hpp:191
#13 storage5 (this=0xa26b3388) at /usr/include/boost/bind/storage.hpp:240
#14 storage6 (this=0xa26b3388) at /usr/include/boost/bind/storage.hpp:289
#15 storage7 (this=0xa26b3388) at /usr/include/boost/bind/storage.hpp:338
#16 list7 (this=0xa26b3388) at /usr/include/boost/bind/bind.hpp:622
---Type <return> to continue, or q <return> to quit---
#17 bind_t (this=0xa26b3378) at /usr/include/boost/bind/bind.hpp:864
#18 manager (op=boost::detail::function::clone_functor_tag, out_buffer=..., 
    in_buffer=...) at /usr/include/boost/function/function_base.hpp:380
#19 manager (op=boost::detail::function::clone_functor_tag, out_buffer=..., 
    in_buffer=...) at /usr/include/boost/function/function_base.hpp:412
#20 boost::detail::function::functor_manager<boost::_bi::bind_t<void, boost::_mfi::mf6<void, dso::EnergyFunctional, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&, Eigen::Matrix<float, 1, 8, 1, 1, 8>*, int, int, Eigen::Matrix<double, 10, 1, 0, 10, 1>*, int>, boost::_bi::list7<boost::_bi::value<dso::EnergyFunctional*>, boost::_bi::value<Eigen::Matrix<float, 4, 1, 0, 4, 1> >, boost::_bi::value<Eigen::Matrix<float, 1, 8, 1, 1, 8>*>, boost::arg<1>, boost::arg<2>, boost::arg<3>, boost::arg<4> > > >::manage (
    in_buffer=..., out_buffer=..., op=boost::detail::function::clone_functor_tag)
    at /usr/include/boost/function/function_base.hpp:440
#21 0x080b51b3 in assign_to_own (f=..., this=0xa63924c0)
    at /usr/include/boost/function/function_template.hpp:909
#22 function4 (f=..., this=0xa63924c0)
    at /usr/include/boost/function/function_template.hpp:749
#23 function (f=..., this=0xa63924c0)
    at /usr/include/boost/function/function_template.hpp:1089
#24 operator= (f=..., this=0x8257b84)
    at /usr/include/boost/function/function_template.hpp:1101
#25 dso::IndexThreadReduce<Eigen::Matrix<double, 10, 1, 0, 10, 1> >::reduce(boost::function<void (int, int, Eigen::Matrix<double, 10, 1, 0, 10, 1>*, int)>, int, int, int)
    (this=<optimized out>, callPerIndex=..., first=<optimized out>, first@entry=0, 
    end=<optimized out>, stepSize=stepSize@entry=50)
    at /media/oem/sda7/DSO/dso-devel/src/util/IndexThreadReduce.h:96
#26 0x080e8f49 in dso::EnergyFunctional::resubstituteF_MT (
    this=this@entry=0x825b040, x=..., HCalib=HCalib@entry=0x8257070, 
    MT=MT@entry=true)
    at /media/oem/sda7/DSO/dso-devel/src/OptimizationBackend/EnergyFunctional.cpp:284
#27 0x080f22fb in dso::EnergyFunctional::solveSystemF (this=0x825b040, 
    iteration=iteration@entry=0, lambda=1.0000000000000001e-05, 
    lambda@entry=0.10000000000000001, HCalib=HCalib@entry=0x8257070)
    at /media/oem/sda7/DSO/dso-devel/src/OptimizationBackend/EnergyFunctional.cpp:912
#28 0x080afccf in dso::FullSystem::solveSystem (this=this@entry=0x8257050, 
    iteration=iteration@entry=0, lambda=lambda@entry=0.10000000000000001)
    at /media/oem/sda7/DSO/dso-devel/src/FullSystem/FullSystemOptimize.cpp:616
#29 0x080b148b in dso::FullSystem::optimize (this=this@entry=0x8257050, 
    mnumOptIts=15)
    at /media/oem/sda7/DSO/dso-devel/src/FullSystem/FullSystemOptimize.cpp:477
---Type <return> to continue, or q <return> to quit---
[... cut ...]
#32 0x08090e91 in dso::FullSystem::addActiveFrame (this=0x8257050, 
    image=image@entry=0xa5ac38c0, id=id@entry=7)
    at /media/oem/sda7/DSO/dso-devel/src/FullSystem/FullSystem.cpp:841
#33 0x0807cf55 in __lambda0::operator() (__closure=0x825b1e4)
    at /media/oem/sda7/DSO/dso-devel/src/main_dso_pangolin.cpp:506
#34 0xb7a07d1e in ?? () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#35 0xb7b9df70 in start_thread (arg=0xa6393b40) at pthread_create.c:312
#36 0xb783dbee in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:129

JakobEngel, 2016-12-22:
strange - maybe boost's bind does not go together with aligned eigen types?
can you see if the error still occurs if you set "MT=false;" at the beginning of EnergyFunctional::resubstituteF_MT, such that the single-threaded variant is used?

mah92, 2016-12-23:
Hi, no luck... still dumps...

mah92, 2016-12-23:
by disabling whole the multithreading functionality, it dumps after changing keyframe in a pangolin assertion (line: 439)
#5  0x08125231 in dso::IOWrap::PangolinDSOViewer::publishGraph (
    this=0x83b8640, connectivity=std::map with 2 elements = {...})
    at /home/oem/Desktop/DSO/DSO/dso-devel/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp:439
#6  0x0808fa62 in dso::FullSystem::makeKeyFrame (this=0x823cb60, fh=0xa4397720)
    at /home/oem/Desktop/DSO/DSO/dso-devel/src/FullSystem/FullSystem.cpp:1179
#7  0x0808fdb2 in dso::FullSystem::deliverTrackedFrame (this=0x823cb60, 
    fh=0xa4397720, needKF=true)
    at /home/oem/Desktop/DSO/DSO/dso-devel/src/FullSystem/FullSystem.cpp:926
#8  0x08093d6a in dso::FullSystem::addActiveFrame (this=0x823cb60, 
    image=0xa438b650, id=8)
    at /home/oem/Desktop/DSO/DSO/dso-devel/src/FullSystem/FullSystem.cpp:841
#9  0x0807da8a in <lambda()>::operator()(void) const (__closure=0x81e17e8)
    at /home/oem/Desktop/DSO/DSO/dso-devel/src/main_dso_pangolin.cpp:493

mah92, 2016-12-25:
after disabling multithreading in settings, it goes further, sigsegving here(CoarseTracker.cpp, line: 318):
__m128 u = _mm_load_ps(buf_warped_u+i);

JakobEngel, 2016-12-25:
can you try the new <devel> branch? 
I fixed the long->uint64_t issue, as well as the missing alignment in CoarseTracker (anyone know of a non-hacky way to 16-byte allocate aligned raw float arrays?).
Unfortunately that may brake compatibility with old <Output3DWrapper> implementations, so for now it's in the devel branch only.

mah92, 2016-12-26:
I give you my best wishes, It worked! God saves you;-)

mah92, 2016-12-26:
On my first old 32bit Dell system, I withdrew a sissegv by disabling "show3d" option, other than this problem everything works...

changkaizhao, 2017-01-02:
I folked the latest  devel branch. Still get the error like this:(64-bit Ubuntu)
[image]

vpicaver, 2017-01-06:
I get the same issue, but with a different back trace:

#0  0x00007ffff5a7c418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff5a7e01a in __GI_abort () at abort.c:89
#2  0x00007ffff5a74bd7 in __assert_fail_base (fmt=<optimized out>, 
    assertion=assertion@entry=0x506f98 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., file=file@entry=0x506f60 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=line@entry=128, 
    function=function@entry=0x520520 <_ZZN5Eigen8internal11plain_arrayIfLi8ELi0ELi32EEC4EvE19__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = float; int Size = 8; int MatrixOrArrayOptions = 0]") at assert.c:92
#3  0x00007ffff5a74c82 in __GI___assert_fail (
    assertion=assertion@entry=0x506f98 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., file=file@entry=0x506f60 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=line@entry=128, 
    function=function@entry=0x520520 <_ZZN5Eigen8internal11plain_arrayIfLi8ELi0ELi32EEC4EvE19__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = float; int Size = 8; int MatrixOrArrayOptions = 0]") at assert.c:101
#4  0x00000000004992d6 in Eigen::internal::plain_array<float, 8, 0, 32>::plain_array (this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128
#5  Eigen::DenseStorage<float, 8, 8, 1, 0>::DenseStorage (this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:187
#6  Eigen::PlainObjectBase<Eigen::Matrix<float, 8, 1, 0, 8, 1> >::PlainObjectBase (this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:457
#7  Eigen::Matrix<float, 8, 1, 0, 8, 1>::Matrix (this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/Matrix.h:259
#8  dso::CoarseInitializer::calcResAndGS (this=this@entry=0x34afb40, lvl=lvl@entry=3, H_out=..., b_out=..., H_out_sc=..., b_out_sc=..., refToNew=..., refToNew_aff=..., 
    plot=false) at /home/philip/Projects/dso/src/FullSystem/CoarseInitializer.cpp:366
#9  0x0000000000499603 in dso::CoarseInitializer::trackFrame (this=0x34afb40, newFrameHessian=newFrameHessian@entry=0x7fffcc0151c0, 
    wraps=std::vector of length 1, capacity 1 = {...}) at /home/philip/Projects/dso/src/FullSystem/CoarseInitializer.cpp:128
#10 0x000000000044fd97 in dso::FullSystem::addActiveFrame (this=0x95fda0, image=image@entry=0x7fffcc015140, id=id@entry=1)
    at /home/philip/Projects/dso/src/FullSystem/FullSystem.cpp:836
#11 0x000000000043a06b in <lambda()>::operator()(void) const (__closure=0x95fb68) at /home/philip/Projects/dso/src/main_dso_pangolin.cpp:493
#12 0x00007ffff63e7c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#13 0x00007ffff71df6fa in start_thread (arg=0x7fffd2f7b700) at pthread_create.c:333
#14 0x00007ffff5b4db5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

vpicaver, 2017-01-06:
I reverted back to 2e89f8ff2f6c7e3bee7a64b83750966d55c9b6a5 and it worked for me.

kgreenek, 2017-02-07:
I'm still seeing this issue. Can't build on master or devel.
I'm running Ubuntu 16.04 64-bit.

Here's the backtrace:
    (gdb) bt
#0  0x00007ffff4d92428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff4d9402a in __GI_abort () at abort.c:89
#2  0x00007ffff4d8abd7 in __assert_fail_base (fmt=<optimized out>, 
    assertion=assertion@entry=0x509b40 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., file=file@entry=0x509b08 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=line@entry=128, 
    function=function@entry=0x5231a0 <_ZZN5Eigen8internal11plain_arrayIfLi8ELi0ELi32EEC4EvE19__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = float; int Size = 8; int MatrixOrArrayOptions = 0]") at assert.c:92
#3  0x00007ffff4d8ac82 in __GI___assert_fail (
    assertion=assertion@entry=0x509b40 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., file=file@entry=0x509b08 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=line@entry=128, 
    function=function@entry=0x5231a0 <_ZZN5Eigen8internal11plain_arrayIfLi8ELi0ELi32EEC4EvE19__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = float; int Size = 8; int MatrixOrArrayOptions = 0]") at assert.c:101
#4  0x0000000000498e66 in Eigen::internal::plain_array<float, 8, 0, 32>::plain_array (this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128
#5  Eigen::DenseStorage<float, 8, 8, 1, 0>::DenseStorage (this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:187
#6  Eigen::PlainObjectBase<Eigen::Matrix<float, 8, 1, 0, 8, 1> >::PlainObjectBase (this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:457
#7  Eigen::Matrix<float, 8, 1, 0, 8, 1>::Matrix (this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/Matrix.h:259
#8  dso::CoarseInitializer::calcResAndGS (this=this@entry=0x35173c0, lvl=lvl@entry=3, H_out=..., b_out=..., H_out_sc=..., b_out_sc=..., refToNew=..., refToNew_aff=..., plot=false)
    at /home/kevin/Projects/catkin_ws/src/dso/src/FullSystem/CoarseInitializer.cpp:366
#9  0x0000000000499193 in dso::CoarseInitializer::trackFrame (this=0x35173c0, newFrameHessian=newFrameHessian@entry=0x7fffd0015220, wraps=std::vector of length 1, capacity 1 = {...})
    at /home/kevin/Projects/catkin_ws/src/dso/src/FullSystem/CoarseInitializer.cpp:128
#10 0x0000000000450a69 in dso::FullSystem::addActiveFrame (this=0x9c76e0, image=image@entry=0x7fffd00151a0, id=id@entry=1) at /home/kevin/Projects/catkin_ws/src/dso/src/FullSystem/FullSystem.cpp:836
#11 0x0000000000439fab in <lambda()>::operator()(void) const (__closure=0x80d008) at /home/kevin/Projects/catkin_ws/src/dso/src/main_dso_pangolin.cpp:493
#12 0x00007ffff56fdc80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#13 0x00007ffff71c86ba in start_thread (arg=0x7fffd4b09700) at pthread_create.c:333
#14 0x00007ffff4e6382d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

kgreenek, 2017-02-08:
Is this issue being used to track all Eigen memory alignment issues, or just a particular one?

Woking-34, 2017-02-14:
I have the same or a similar issue on:
- Ubuntu 16.04.2 LTS
- 4.4.0-62-generic #83-Ubuntu SMP Wed Jan 18 14:10:15 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Reading Photometric Calibration from file ../data/tum/sequence_25/pcalib.txt
Reading Vignette Image from ../data/tum/sequence_25/vignette.png
Successfully read photometric calibration!
got 4090 images and 4090 timestamps and 4090 exposures.!
ImageFolderReader: got 4090 files in ../data/tum/sequence_25/images.zip!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
START PANGOLIN!
dso_dataset: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = float; int Size = 8; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Aborted (core dumped)

JakobEngel, 2017-02-15:
I added one more fix to the devel branch that may fix the issue

Woking-34, 2017-02-15:
Great, the issue is gone with your fix, thanks!

wystnphux, 2017-02-17:
Thanks a lot for fixing the issues with Eigen in the devel branch.

JakobEngel, 2017-02-20:
pushed this to master - please re-open if there are any alignment issues left :)

closed 2017-02-20 by JakobEngel