JakobEngel/dso #25: Running on Euroc MAV dataset

andre-nguyen, 2016-12-06:
Hi,

how did you manage to run dso on the Euroc MAV dataset? I was about to attempt but then I realized I wouldn't be able to get the gamma and vignette calibration since they don't provide that kind of calibration data.

Thanks,

andre-nguyen, 2016-12-06:
Oops, I misunderstood the readme. I get it now!

closed 2016-12-06 by andre-nguyen

Yelen719, 2017-01-07:
How was the performance on EuRoC MAV?

andre-nguyen, 2017-01-08:
It was great actually, now if only I knew how to use GTSAM or something to scale the measurements...

447425299, 2017-05-03:
@andre-nguyen  I have run dso on EUROC, but the performance is not very good, Are you sure it is great on EUROC?

andre-nguyen, 2017-05-03:
I used it on one of the machine hall bags and the point cloud was actually pretty nice and the vehicle came back really not for from its starting point. I have no other comparison criteria or performance data to show.

447425299, 2017-05-04:
ok thanks

zhehangT, 2017-12-27:
hi, how did you define the  calibration data of EUROC, please?
Can you give any examples, please?
@andre-nguyen

SkyRiderMike, 2018-04-16:
How can you run this program on EuRoC ? @andre-nguyen

andre-nguyen, 2018-04-17:
This thread is almost a year old, I suggest just reading the Readme carefully, seems like it worked for me and others in the past.

hobbeshunter, 2020-10-15:
I used

```
KannalaBrandt 460.92587684824869 459.56960125805508 365.90627988127127 249.33982146124706 -0.004161020735327023 0.02326805021977024 -0.02673515372903321 0.011272262685694161
752 480
crop
752 480
```

for cam0