JakobEngel/dso #26: Compiling with Boost thread on Mac

kitizz, 2016-12-07:
When compiling the dso_dataset binary on Mac, I encountered the following error:
```
ld: library not found for -lboost_thread
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/dso_dataset] Error 1
make[1]: *** [CMakeFiles/dso_dataset.dir/all] Error 2
make: *** [all] Error 2
```

This was fixed by changing `boost_thread` to `boost_thread-mt` in the following line in CMakeLists.txt:
```
target_link_libraries(dso_dataset dso boost_system boost_thread cxsparse ${LIBZIP_LIBRARY} ${Pangolin_LIBRARIES} ${OpenCV_LIBS})
```

This appears to be Mac and Windows specific. And even on Mac it may be Homebrew specific, so I'm not sure if an OS check in CMakeLists is the correct way to fix it.

I thought I'd put this here in case anyone else searched for the problem =)

kitizz, 2016-12-07:
This was fixed by changing `boost_thread` to `boost_thread-mt` in the following line in CMakeLists.txt:
```
target_link_libraries(dso_dataset dso boost_system boost_thread cxsparse ${LIBZIP_LIBRARY} ${Pangolin_LIBRARIES} ${OpenCV_LIBS})
```

This appears to be Mac and Windows specific. And even on Mac it may be Homebrew specific, so I'm not sure if an OS check in CMakeLists is the correct way to fix it.

I thought I'd put this here in case anyone else searched for the problem =)

JakobEngel, 2016-12-09:
thx!  I'm planning to remove the boost dependency (I'm only using it for thread / mutex stuff) and replace it with the c++11 equivalents when I have time, that should also resolve the issue.

closed 2016-12-12 by JakobEngel

onurays, 2017-01-13:
Any progress, plan or workaround on this issue?

iamazy, 2019-08-17:
> This was fixed by changing `boost_thread` to `boost_thread-mt` in the following line in CMakeLists.txt:
>
> ```
> target_link_libraries(dso_dataset dso boost_system boost_thread cxsparse [...]

thanks,It solved my problem

PR #98 (2017-10-18): Compile on MacOS 
fix #26

Also I have problem with code compile. Fixed by changing type of `connectivityMap` to:

    std::map<uint64_t,Eigen::Vector2i> connectivityMap;
commit 94335ce6 (2017-10-18): Update CMakeLists.txt

Compile on MacOS 
fix #26

Also I have problem with code compile. Fixed by changing type of `connectivityMap` to:

    std::map<uint64_t,Eigen::Vector2i> connectivityMap;