JakobEngel/dso #28: Compile Error

mah92, 2016-12-16:
Hi
I encountered an error in compilation. I work with ubuntu 16.04, gcc 5.4, boost 1.58.0, opencv 2.4.9.1
I installed Pangolin as mensioned in the documentation successfully.
here is the error(and warnings if important):
/home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp: In member function ‘dso::Vec3f dso::CoarseInitializer::calcResAndGS(int, dso::Mat88f&, dso::Vec8f&, dso::Mat88f&, dso::Vec8f&, dso::SE3, dso::AffLight, bool)’:
/home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:469:18: warning: SSE vector return without SSE enabled changes the ABI [-Wpsabi]
    acc9.updateSSE(
                  ^
In file included from /usr/lib/gcc/i686-linux-gnu/5/include/x86intrin.h:31:0,
                 from /usr/include/i386-linux-gnu/c++/5/bits/opt_random.h:33,
                 from /usr/include/c++/5/random:50,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/eigen3/Eigen/Core:232,
                 from /home/oem/Desktop/DSO/DSO/dso/src/util/NumType.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/usr/lib/gcc/i686-linux-gnu/5/include/xmmintrin.h: In member function ‘void dso::Accumulator11::shiftUp(bool)’:
/usr/lib/gcc/i686-linux-gnu/5/include/xmmintrin.h:971:1: error: inlining failed in call to always_inline ‘void _mm_store_ps(float*, __m128)’: target specific option mismatch
 _mm_store_ps (float *__P, __m128 __A)
 ^
In file included from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:28:0,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/home/oem/Desktop/DSO/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:157:85: error: called from here
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                                                                                     ^
In file included from /usr/lib/gcc/i686-linux-gnu/5/include/x86intrin.h:31:0,
                 from /usr/include/i386-linux-gnu/c++/5/bits/opt_random.h:33,
                 from /usr/include/c++/5/random:50,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/eigen3/Eigen/Core:232,
                 from /home/oem/Desktop/DSO/DSO/dso/src/util/NumType.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/usr/lib/gcc/i686-linux-gnu/5/include/xmmintrin.h:181:1: error: inlining failed in call to always_inline ‘__m128 _mm_add_ps(__m128, __m128)’: target specific option mismatch
 _mm_add_ps (__m128 __A, __m128 __B)
 ^
In file included from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:28:0,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/home/oem/Desktop/DSO/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:157:17: error: called from here
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                 ^
In file included from /usr/lib/gcc/i686-linux-gnu/5/include/x86intrin.h:31:0,
                 from /usr/include/i386-linux-gnu/c++/5/bits/opt_random.h:33,
                 from /usr/include/c++/5/random:50,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/eigen3/Eigen/Core:232,
                 from /home/oem/Desktop/DSO/DSO/dso/src/util/NumType.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/usr/lib/gcc/i686-linux-gnu/5/include/xmmintrin.h:922:1: error: inlining failed in call to always_inline ‘__m128 _mm_load_ps(const float*)’: target specific option mismatch
 _mm_load_ps (float const *__P)
 ^
In file included from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:28:0,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/home/oem/Desktop/DSO/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:157:17: error: called from here
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                 ^
In file included from /usr/lib/gcc/i686-linux-gnu/5/include/x86intrin.h:31:0,
                 from /usr/include/i386-linux-gnu/c++/5/bits/opt_random.h:33,
                 from /usr/include/c++/5/random:50,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/eigen3/Eigen/Core:232,
                 from /home/oem/Desktop/DSO/DSO/dso/src/util/NumType.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/usr/lib/gcc/i686-linux-gnu/5/include/xmmintrin.h:922:1: error: inlining failed in call to always_inline ‘__m128 _mm_load_ps(const float*)’: target specific option mismatch
[... 52 lines cut ...]
 ^
In file included from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:28:0,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/home/oem/Desktop/DSO/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:164:17: error: called from here
     _mm_store_ps(SSEData1m, _mm_add_ps(_mm_load_ps(SSEData1k),_mm_load_ps(SSEData1m)));
                 ^
In file included from /usr/lib/gcc/i686-linux-gnu/5/include/x86intrin.h:31:0,
                 from /usr/include/i386-linux-gnu/c++/5/bits/opt_random.h:33,
                 from /usr/include/c++/5/random:50,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/include/eigen3/Eigen/Core:232,
                 from /home/oem/Desktop/DSO/DSO/dso/src/util/NumType.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:27,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/usr/lib/gcc/i686-linux-gnu/5/include/xmmintrin.h:922:1: error: inlining failed in call to always_inline ‘__m128 _mm_load_ps(const float*)’: target specific option mismatch
 _mm_load_ps (float const *__P)
 ^
In file included from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.h:28:0,
                 from /home/oem/Desktop/DSO/DSO/dso/src/FullSystem/CoarseInitializer.cpp:32:
/home/oem/Desktop/DSO/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:164:17: error: called from here
     _mm_store_ps(SSEData1m, _mm_add_ps(_mm_load_ps(SSEData1k),_mm_load_ps(SSEData1m)));
                 ^
make[2]: *** [CMakeFiles/dso.dir/src/FullSystem/CoarseInitializer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

mah92, 2016-12-16:
And my system is a dual-core 32bit Intel T2300

JakobEngel, 2016-12-19:
humm - seems to have something to do with SSE and the 32bit CPU. 
can you try adding something like "-msse4.1" to the compiler flags (here: https://github.com/JakobEngel/dso/blob/master/CMakeLists.txt#L31)

mah92, 2016-12-20:
Excuse me for my late answer, both switching to:
"${SSE_FLAGS} -O3 -g -std=c++0x -fno-omit-frame-pointer -msse4.1"
or
"${SSE_FLAGS} -O3 -g -std=c++0x -fno-omit-frame-pointer -march=native"
solved the problem.
Thanks

closed 2016-12-20 by mah92