JakobEngel/dso #30: Compile Error2

mah92, 2016-12-18:
In the name of God
Hi, After unsuccessfully trying to compile the lib on my system, I tried it on another one(64 bits, Ubuntu14.04, gcc 4.8.4) but got this error log:
make 
[  4%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o
In file included from /media/oem/sda7/DSO/dso/src/FullSystem/CoarseTracker.h:32:0,
                 from /media/oem/sda7/DSO/dso/src/FullSystem/FullSystem.cpp:47:
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:121:11: error: ‘__m128’ does not name a type
     const __m128 val)
           ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:121:18: error: ISO C++ forbids declaration of ‘val’ with no type [-fpermissive]
     const __m128 val)
                  ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:137:11: error: ‘__m128’ does not name a type
     const __m128 val)
           ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:137:18: error: ISO C++ forbids declaration of ‘val’ with no type [-fpermissive]
     const __m128 val)
                  ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h: In member function ‘void dso::Accumulator11::updateSSE(int)’:
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:123:56: error: ‘_mm_load_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                        ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:123:61: error: ‘_mm_add_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                             ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:123:62: error: ‘_mm_store_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                              ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h: In member function ‘void dso::Accumulator11::updateSSENoShift(int)’:
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:139:56: error: ‘_mm_load_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                        ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:139:61: error: ‘_mm_add_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                             ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:139:62: error: ‘_mm_store_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                              ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h: In member function ‘void dso::Accumulator11::shiftUp(bool)’:
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:157:59: error: ‘_mm_load_ps’ was not declared in this scope
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                                                           ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:157:83: error: ‘_mm_add_ps’ was not declared in this scope
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                                                                                   ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:157:84: error: ‘_mm_store_ps’ was not declared in this scope
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                                                                                    ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:164:61: error: ‘_mm_load_ps’ was not declared in this scope
     _mm_store_ps(SSEData1m, _mm_add_ps(_mm_load_ps(SSEData1k),_mm_load_ps(SSEData1m)));
                                                             ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:164:85: error: ‘_mm_add_ps’ was not declared in this scope
     _mm_store_ps(SSEData1m, _mm_add_ps(_mm_load_ps(SSEData1k),_mm_load_ps(SSEData1m)));
                                                                                     ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:164:86: error: ‘_mm_store_ps’ was not declared in this scope
     _mm_store_ps(SSEData1m, _mm_add_ps(_mm_load_ps(SSEData1k),_mm_load_ps(SSEData1m)));
                                                                                      ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h: At global scope:
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:283:11: error: ‘__m128’ does not name a type
     const __m128 J0,const __m128 J1,
           ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:283:18: error: ISO C++ forbids declaration of ‘J0’ with no type [-fpermissive]
     const __m128 J0,const __m128 J1,
                  ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:283:27: error: ‘__m128’ does not name a type
     const __m128 J0,const __m128 J1,
                           ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:283:34: error: ISO C++ forbids declaration of ‘J1’ with no type [-fpermissive]
     const __m128 J0,const __m128 J1,
                                  ^
[... 354 lines cut ...]
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:1158:59: error: ‘J8w’ was not declared in this scope
    _mm_store_ps(pt, _mm_add_ps(_mm_load_ps(pt),_mm_mul_ps(J8w,J8))); pt+=4;
                                                           ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h: In member function ‘void dso::Accumulator9::shiftUp(bool)’:
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:1327:68: error: ‘_mm_load_ps’ was not declared in this scope
      _mm_store_ps(SSEData1k+4*i, _mm_add_ps(_mm_load_ps(SSEData+4*i),_mm_load_ps(SSEData1k+4*i)));
                                                                    ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:1327:96: error: ‘_mm_add_ps’ was not declared in this scope
      _mm_store_ps(SSEData1k+4*i, _mm_add_ps(_mm_load_ps(SSEData+4*i),_mm_load_ps(SSEData1k+4*i)));
                                                                                                ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:1327:97: error: ‘_mm_store_ps’ was not declared in this scope
      _mm_store_ps(SSEData1k+4*i, _mm_add_ps(_mm_load_ps(SSEData+4*i),_mm_load_ps(SSEData1k+4*i)));
                                                                                                 ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:1336:70: error: ‘_mm_load_ps’ was not declared in this scope
      _mm_store_ps(SSEData1m+4*i, _mm_add_ps(_mm_load_ps(SSEData1k+4*i),_mm_load_ps(SSEData1m+4*i)));
                                                                      ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:1336:98: error: ‘_mm_add_ps’ was not declared in this scope
      _mm_store_ps(SSEData1m+4*i, _mm_add_ps(_mm_load_ps(SSEData1k+4*i),_mm_load_ps(SSEData1m+4*i)));
                                                                                                  ^
/media/oem/sda7/DSO/dso/src/OptimizationBackend/MatrixAccumulators.h:1336:99: error: ‘_mm_store_ps’ was not declared in this scope
      _mm_store_ps(SSEData1m+4*i, _mm_add_ps(_mm_load_ps(SSEData1k+4*i),_mm_load_ps(SSEData1m+4*i)));
                                                                                                   ^
make[2]: *** [CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o] Error 1
make[1]: *** [CMakeFiles/dso.dir/all] Error 2
make: *** [all] Error 2

rongbohou, 2016-12-19:
I have got this problem too.After I modified the CMakeLIsts.txt like below,I can compile the program.

# flags
add_definitions("-DENABLE_SSE")
set(CMAKE_CXX_FLAGS
   "${CMAKE_CXX_FLAGS} -O3 -g -std=c++0x -fno-omit-frame-pointer -march=native"
#   "${SSE_FLAGS} -O3 -g -std=c++0x -fno-omit-frame-pointer"
)

mah92, 2016-12-19:
Thanks, that solved it :-)

closed 2016-12-19 by mah92

mah92, 2016-12-21:
Just one more note: I made a mistake on that my os was actually 32 bit... Compiler error was gone but another problem mensioned as eigen assertation error was raised...