JakobEngel/dso #31: Compiling DSO on ARM processor

sunghoon031, 2016-12-24:
I found that you use SSE optimization, which is not supported in ARM devices...

Do you have any ideas/suggestions for workaround if I would like to compile it on ARM devices (e.g. Odroid)? 

BTW, Thank a lot for sharing your source code! It's super cool !! :+1:

JakobEngel, 2016-12-25:
Yes - I removed the respective non-SSE versions of the functions where SSE is used,
 since I didn't want to always have to maintain / adapt two versions during development.
However, it should be fairly straight-forward to re-introduce them, especially since SSE is only
used in a fairly limited set of functions.
If you need the performance (in particular for the CoarseTracker, where it is probably most relevant), you can use NEON instead.

hwiryong-jung, 2016-12-27:
On NVIDIA's TX1, same issue exists.. (it's also ARM processor : (|).....  )

closed 2016-12-27 by sunghoon031

adroit91, 2017-03-29:
Hi,

I have successfully been able to use it by wrapping with:
https://github.com/jratcliff63367/sse2neon

israelshirk, 2017-05-22:
Repeat from #57: Here's a site with some info on converting the Intel SSE instructions to ARM NEON instructions: http://codesuppository.blogspot.com/2015/02/sse2neonh-porting-guide-and-header-file.html, which is referenced above by @adroit91 for the source code only.

I'll likely throw an automated script to do this translation for me in the next day or two, but it's easy enough off the blog post and source.