JakobEngel/dso #34: kitti vo dataset seq00 result

hwiryong-jung, 2016-12-26:
[image]

JakobEngel, 2016-12-26:
In general, DSO does not work very well on the KITTI dataset because the distortion calibration used to pre-rectify the images for the dataset was rather poor -- and since there is no access to the original raw images (or calibration datasets), this is hard to undo (e.g. using on-line calibration).
Poor calibration in turn manifests as small geometric distortions in the images ("geometric noise") - to which direct methods are much more sensitive (see the geometric noise vs. photometric noise experiment in the DSO paper).

Furthermore, I believe the above picture of ORB-SLAM is "with scale correction" (scale drift being by far the most significant source of error), the original figure in fact looks like this:
[image]
(see http://ieeexplore.ieee.org/document/7535431/)

to answer your question, you may want to keep the original aspect ratio, but down-scale the images to roughly VGA resolution (300k pixel)... e.g.
1241 376 
crop
960 288

closed 2017-01-10 by JakobEngel

kozuch, 2017-04-18:
@JakobEngel 

> there is no access to the original raw images (or calibration datasets)

I think this is not true - all raw KITTI data is available on [this page](http://www.cvlibs.net/datasets/kitti/raw_data.php) - the odometry dataset (which you are probably talking about) is only a subset of their raw dataset. Both the unrectified images and calibration images are available (see the linked page in detail). However, the question is whether their calibration images are sufficient actually (they seem to have used very few checkerboards).

There also is "conversion table" from the odometry sequence numbering to the raw scene numbering in the readme of [odometry devkit zip file](http://kitti.is.tue.mpg.de/kitti/devkit_odometry.zip):

Odometry Nr.     Raw sequence name     Start   End
00: 2011_10_03_drive_0027 000000 004540
01: 2011_10_03_drive_0042 000000 001100
02: 2011_10_03_drive_0034 000000 004660
03: 2011_09_26_drive_0067 000000 000800
04: 2011_09_30_drive_0016 000000 000270
05: 2011_09_30_drive_0018 000000 002760
06: 2011_09_30_drive_0020 000000 001100
07: 2011_09_30_drive_0027 000000 001100
08: 2011_09_30_drive_0028 001100 005170
09: 2011_09_30_drive_0033 000000 001590
10: 2011_09_30_drive_0034 000000 001200 

Do you think this will help for the KITTI sequences to work better with DSO? Can you try to look at the unrectified and calibration images to judge? I would appreciate that since I would like to test DSO on the KITTI dataset.

jimaldon, 2019-08-20:
Any luck on this? The KITTI website is down.

mrgransky, 2019-10-19:
03: 2011_09_26_drive_0067 is not found in the KITTI benchmard dataset.

both of the following links are broken:

https://s3.eu-central-1.amazonaws.com/avg-kitti/raw_data/2011_09_26_drive_0067/2011_09_26_drive_0067_sync.zip

http://kitti.is.tue.mpg.de/kitti/raw_data/2011_09_26_drive_0067/2011_09_26_drive_0067_sync.zip

Has this seq (03) somehow removed ?