JakobEngel/dso #35: API document

gogojjh, 2016-12-28:
Will you provide the additive API document of this code?

JakobEngel, 2016-12-28:
To be honest, probably not. The closest to an API is the documentation in 
https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/Output3DWrapper.h
and the example in 
https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/OutputWrapper/SampleOutputWrapper.h

That should be enough to figure out how to use the code, and integrate it into your own project. 
If you want to go deeper, you will have to figure out what does what yourself, the paper ( http://vision.in.tum.de/_media/spezial/bib/engel2016dso.pdf ) should help with that. 

It should be mentioned that the code is the result of a fairly long-lasting research project, during the course of which it has been re-written and re-factored many many times ... Thus, from a "software-engineering" standpoint it is probably quite terrible ;).

mah92, 2016-12-28:
Hi, I had some questions, what is the CPAR variable set to 4?

JakobEngel, 2017-01-02:
CPAR is the number of camera intrinsic parameters that are calibrated on-line (fx, fy, cx, cy by default) - I have however never tested any other number. 
The parameters in settings.cpp that do not have any description can mostly be kept as they are, you should not have to change them for anything really - I just have a habit of collecting all possible parameters / settings in one central place.

s3r637, 2017-04-20:
The initialization part (CoarseInitializer) differs from the tracking part (CoarseTracker) wasn't mentioned in the paper, right? The paper describes only the part which continuously update the existing motion model.

NikolausDemmel, 2017-12-07:
Closing since the question seems answered / due to lack of activity.

closed 2017-12-07 by NikolausDemmel