JakobEngel/dso #36: How to add imu to solve monocular scale problem

xiaopenghuang, 2017-01-05:
Thanks for your great work. I want to add imu to solve monocular scale problem. But I donot how to use it , only use imu preintergation (between two image ) to predict next image alignment position .
Can you give me any clue or advice or how to combine dso with imu ,or any relevant paper you think might help. Thank you very much.

MichaelGrupp, 2017-02-03:
Go to google scholar and search for "visual inertial odometry", there are many approaches.

It's definitely not a trivial task!

syurkevi, 2017-02-07:
In terms of this project, which classes would need to be touched? 
Would it be enough to estimate the pose during the creation of new FrameHessians in trackNewCoarse() or would the residuals used in the optimizer also need to be changed?

fluentart, 2017-09-08:
I dont think an IMU will help with scale. Best would be to use some kind of range sensor (ultrasonic, or laser) and use that reading to set scale.

closed 2017-09-09 by xiaopenghuang

tibitomabraham, 2017-10-26:
@xiaopenghuang I am also trying to solve the scale issue of DSO. Could you find a solution?