JakobEngel/dso #39: Error building dso-slam with Visual Studio

wystnphux, 2017-02-13:
I get an error building dso-slam with Visual Studio 2015. The first error message is as follows:

>Error	C2976	'std::deque': too few template arguments	dso	C:\XXX\dso-slam\dso\src\FullSystem\FullSystem.h	310	

Line 310 of FullSystem.h:
`std::deque<FrameHessian*> unmappedTrackedFrames;`

Is this related to the the definition of `struct FrameHessian` in HessianBlacks.h?

Actually, I'd like to fix this issue but I don't know where to start searching. Maybe there is someone more experienced that may give a hint on this. What would be useful information to look at?

Kind regards

wystnphux, 2017-02-16:
An `#include <deque>` above all other includes did solve the problem.

JakobEngel, 2017-02-20:
can you create a pull request with the added includes? or let me know where exactly you had to add them? thanks!

fszeliga, 2017-04-18:
@wystnphux hey, did you have to do anything special to run dso under visual studio? I have a weird unresolved symbol when initializing FullSystem (other components of dso_dataset work just fine)...

NikolausDemmel, 2017-12-07:
Closing since the question seems answered / due to lack of activity.

closed 2017-12-07 by NikolausDemmel