JakobEngel/dso #40: Can not declared "EIGEN_ALIGN32"

zqp2009happy, 2017-02-20:
When I make dso, it has errors about not declared EIGEN_ALIGN32 in FullSystem/CoarseInitializer.cpp. How to fix this? Thank you.

zhudi0609, 2017-02-20:
Same errors occur when I try to make both in master and devel. 
Any ideas? Thanks.

zhudi0609, 2017-02-20:
Same errors occur when I try to make dso in ubuntu 16.04.
Any ideas? Thanks.

ticolas121, 2017-02-21:
Same issue for me

zqp2009happy, 2017-02-21:
It's good for the old version(15, Feb), but when I updated to the latest version(20, Feb), it reported the issue.

mthz, 2017-02-21:
EIGEN_ALIGN32 was introduced in eigen 3.3; therefore not available with default eigen version installed.
Workaround: define yourself 
#define EIGEN_ALIGN32 EIGEN_ALIGN_TO_BOUNDARY(32)
or use directly
EIGEN_ALIGN_TO_BOUNDARY(32)   VecNRf dp0;

zqp2009happy, 2017-02-22:
Thanks. I downloaded the newest eigen, and it worked.

Cakem1x, 2017-02-22:
Had the same issue like OP and used the same workaround mthz mentioned:
I replaced all ocurrences of EIGEN_ALIGN32 with EIGEN_ALIGN_TO_BOUNDARY(32)
Building works fine now.

Buuut when I run the dso_dataset binary, I get an "UnalignedArrayAssert" failure from Eigen, which is explained here:
http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html
Detailed error below [1].
Does anyone of you get the same error or can report that the dso_dataset binary runs fine on the current master with a given Eigen version (which one?)?

I'm using Eigen3.2 on an Ubuntu 14.04.

[1]: Error thrown after less than a second runtime:
`dso_dataset: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:78: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = float; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.`

lowest gdb backtrace frame which is in the dso code:
`dso::EnergyFunctional::resubstituteF_MT (this=this@entry=0x83920c0, x=..., HCalib=HCalib@entry=0x823ed80, MT=MT@entry=true) at dso/src/OptimizationBackend/EnergyFunctional.cpp:282`