JakobEngel/dso #41: Eigen assert fails: UnalignedArrayAssert

Cakem1x, 2017-02-22:
I'm using Eigen3.2 on an Ubuntu 14.04 32bit system.

First, I had the issue discussed here:
https://github.com/JakobEngel/dso/issues/40

I worked around that issue by replacing EIGEN_ALIGN32 with EIGEN_ALIGN_TO_BOUNDARY(32).

Building works fine now, but when I run the dso_dataset binary, I get an "UnalignedArrayAssert" failure from Eigen, which is explained here:
http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html
Detailed error below [1].
Does anyone get the same error or can report that the dso_dataset binary runs fine on the current master with a given Eigen version (which one?)?

I tried rolling back to the december version, but that didn't change anything.

I tested it on the sequence_09 and sequence_11 datasets.

[1]: Error thrown after less than a second runtime:
dso_dataset: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:78: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = float; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion (reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

lowest gdb backtrace frame which is in the dso code:
dso::EnergyFunctional::resubstituteF_MT (this=this@entry=0x83920c0, x=..., HCalib=HCalib@entry=0x823ed80, MT=MT@entry=true) at dso/src/OptimizationBackend/EnergyFunctional.cpp:282

Complete stacktrace here: http://pastebin.com/0vzLQGQR

Cakem1x, 2017-02-23:
I managed to fix the issue, at least with my setup.

The problem seems to come from the boost::bind functionality used in the EnergyFunctional::resubstituteF_MT method to bind the EnergyFunctional::resubstituteFPt function for the reduce-call on IndexThreadReduce.

Eigen always wants its fixed-size Types to be passed by reference, to avoid misalignment. (see http://eigen.tuxfamily.org/dox-devel/group__TopicPassingByValue.html)
And boost::bind always internally passes bound function parameters by-value, which probably caused the issue I had. (see http://stackoverflow.com/questions/10658062/does-using-eigen-types-with-boostbind-automatically-violate-eigens-only-pass)
By using boost::ref or boost::cref (cref for const references), this can be circumvented.

I'll update my pull-request to also add this change.

closed 2017-02-23 by Cakem1x

xxxxxiaoming, 2017-03-24:
Hi! I have met the same error when I run the dso_dataset.
The way you solve this problem is that replacing boost::bind with boost::ref in EnergyFunctional::resubstituteF_MT?

Cakem1x, 2017-03-24:
Not exactly; I used boost::ref on one of boost::bind's parameters.
Have a look at the changes in the second commit of my pull request, to see the changes in code:
https://github.com/JakobEngel/dso/pull/43/commits/3dee169b47b76dd83e75fedf38e0b83ca74bfae7

xxxxxiaoming, 2017-03-24:
Thanks! It really works!