JakobEngel/dso #47: Very low FPS in Visual Studio

lshGame, 2017-03-09:
Hey guys. I managed to compile DSO with visual studio2013, but DSO(release) ran at a very low FPS( 4 fps after initialization).
While compiling the release version of DSO, visual studio stuck in "generating code" with Optimization /O2 on. SO I disabled optimization. Is there any way to turn on /O2?
by the way, My CPU is i7-3610QM.

JakobEngel, 2017-03-11:
I myself have never used Visual Studio, so I cannot answer your question. However, I can confirm that when compiling without full optimization (or even in debug mode), DSO will be very slow, in particular due to the heavy use of Eigen.

lshGame, 2017-03-16:
Thanks for your reply. My problem is solved. The cause of the issue is that I used Eigen3 which was not compiled. Now, DSO runs at 20FPS(tum dataset) in windows. Awesome work!

Lucky8Young, 2017-03-20:
@lshGame I'm very sorry to bother you, friend. I want to compile DSO with vs2013, but I don't know how to  
do it successfully. I have tried several times but all failed. Could you please tell me how to do it or give some advices? Thank you very much!

jhincapie, 2017-03-27:
I am also interested in compiling with VS, but have not been successful. Anyone knows the steps?

YJCITA, 2017-09-14:
@lshGame  I'm very sorry to bother you. But I wonder what's "I used Eigen3 which was not compiled" mean? As I know, Eigen3 is header files?

lshGame, 2017-09-14:
@YJCITA You're right that Eigen3 is header files, which means it's not necessary to compile it. But if you met the same low FPS problem in windows, you can try it. It may work

lshGame, 2017-09-14:
@Lucky8Young @jhincapie @fszeliga I'm terribly sorry to reply late. If you still want to compile dso in windows with VS2013, you can refer to this article. https://mp.weixin.qq.com/s/5k_lX7Dof65AzI7uBC9HUg

NikolausDemmel, 2017-12-07:
Closing since the question seems answered / due to lack of activity.

closed 2017-12-07 by NikolausDemmel