JakobEngel/dso #51: How to get camera pose for every frame in KITTI dataset using DSO_SLAM?

ChidanandKumarKS, 2017-04-10:
How to get camera pose for every frame in KITTI dataset using DSO_SLAM?
Please provide me necessary inputs...

Regards
Kumar

s3r637, 2017-04-20:
Hi Kumar,

pass just `true` as second parameter in line [900](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/FullSystem/FullSystem.cpp#L900). This change causes that every frame becomes a keyframe. The callback function [`virtual void publishKeyframes(...)`](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/IOWrapper/OutputWrapper/SampleOutputWrapper.h#L77) will called for every frame (I think except some first initialization frames). Make sure that they are `final = true`.

Change the `benchmarkSetting_width` and `benchmarkSetting_height` if you do not want to use the fast settings, e.g
`benchmarkSetting_width = 1232; benchmarkSetting_height = 368;`
add this line [here](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/main_dso_pangolin.cpp#L116).

This will slow down it and probably it doesn't run in real-time anymore.

Hope this helps
Sergej

wow2006, 2017-05-07:
Hello @s3r637 ,
I want to get Camera Matrix [R, T] at the world coordinates ie. relative to (0, 0) first camera and translations is relative to (0, 0) not last frame. When I write camera position using camToWorld at publishKeyframes. I got relative translation and rotation. I compared it with Bundle adjustment output and I found out that the two (bundle, DSO) have different translation and rotation for all images.

this my code to write camera at [publishKeyframes](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/IOWrapper/OutputWrapper/SampleOutputWrapper.h#L77):
```cpp
if (frame->poseValid)
{
    boost::format mFormat("%0.5f %0.5f %0.5f\n");
    std::ofstream mLogging("camera.out");
    const auto& R = frame->camToWorld.rotationMatrix().transpose();
    const auto& T = frame->camToWorld.translation();
    mLogging << mFormat % R(0, 0) % R(0, 1) % R(0, 2)
                     << mFormat % R(1, 0) % R(1, 1) % R(1, 2)
                     << mFormat % R(2, 0) % R(2, 1) % R(2, 2)
                     << mFormat % T(0) % T(1) % T(2);
    mLogging.flush();
}
```

Could you please tell me what I am missing?!

s3r637, 2017-05-08:
Hi @wow2006,

I don't get what you mean with "bundle adjustment". I assume that you run another algorithm to produce equivalent results. If so, you are right! They produce different translation and rotation for all images and that's absolutely fine. This behavior is linked to various factors (e.g. scale factor).
You should plot/visualize your results to see what is going on. Both plots should look equivalent (except scale factor and scale drift).

Nevertheless, make sure that the frames are `final = true`.
```
if (final) {
    ... // your code
}
```
Otherwise, your code looks fine.

cheers

wow2006, 2017-05-08:
Hi @s3r637,

Thanks for your quick response. Bundle adjustment is another structure from motion technique which builds 3D geometry from not sequential images. All I know that bundle output is on global coordinate ie. relative to the first frame. I want to transform DSO coordinates (relative to the previous frame) to global (absolute) coordinates.

Best regards

s3r637, 2017-05-08:
Hi @wow2006,

You are right about bundle adjustment (my assumption was right) but DSO also already provides global coordinates (`camToWorld` needs no transformation!).

cheers

vannem95, 2017-05-11:
@s3r637 Hey Sergej, I am new to dso. My goal is to output camera pose in real-time. I looked into dso-ros. The calibrations didn't work but i found someone else's pcalib and vignette which are kinda working. The changes you mentioned in comment#2, do they output anything when done with dso-ros? what do suggest i do to get the camera pose outputted when dso-ros ing? Thanks in advance

s3r637, 2017-05-11:
Hey @vannem95,

I recommend you to use pcalib from one of this [datasets](https://vision.in.tum.de/mono-dataset) (i.g. sequence_14) and create your own vignette (i.e. with GIMP). Set pixels

- white: which should be observed
- black: which (are visible in all frames i.e. car bonnet and) should not be considered

The (`calib` parameter) geometric camera calibration file should contains the projection/camera matrix.
Furthermore, `dso_ros` should subscribe on the topic with rectified images.

Then `dso_ros` should behaves identical.

cheers

wow2006, 2017-05-18:
Hello @s3r637,

I really appreciate your response and patience. 
1. After reading more about DSO and bundle. I want to know ***"Is DSO origin point (0, 0, 0) is first displayed frame or the first frame dropped?"*** ie. In my case, during Initiations DSO drop a couple of frames and start displaying at 7th frame.
3. What is coordinate system of DSO?
2. What is the order of translation matrix extracted from `camToWorld`  "R*T or T*R"?!

Best regards

s3r637, 2017-05-19:
Hi @wow2006,

As I understand:

1. If the first initialization failed another attempt is started until initialization is succeed or no more frames are available. The initialization tries to detect changes (motion) between incoming frames. If motion (between first and second frame) is detected it waits six more frames. That mean, it takes eight (first) frames for initialization (and that only if between first and second frame changes are detected, otherwise it takes more frames). Then initialization happens between the first frame (`incoming_id = 0`) and the eighth frame (`incoming_id = 7`). I'm not sure if all other frames `[1-6]` are used (but it seems so) for the initialization proposal. Anyway, you will never obtain a pose for this (`[1-6]`) frames. This behavior is changeable (see [src/FullSystem/CoarseInitializer.cpp#L271](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/FullSystem/CoarseInitializer.cpp#L271)) but then initialization may failed (to often). (@ChidanandKumarKS Together with  https://github.com/JakobEngel/dso/issues/51#issuecomment-295774219 I'm able to receive poses for all frames except for the last `dso::setting_maxFrames = 7` frames, because they are not marginalized).
**The first frame (isn't dropped! and) lies (approx) in origin point (0, 0, 0).**
2. It is a cartesian coordinate system. (At least how the pangolin gui interpreted the data.)
3. Usually `RT` but it depends on how you extract it. `camToWorld`  is from type `SE3Group<double>` and contains the _translation vector_ and _unit quaternion_.
But `auto const rt = camToWorld.matrix3x4();` provides you an `3x4-[R|T]` matrix.

Hope this helps

wow2006, 2017-06-26:
Hi @s3r637

Thanks a lot for your reply. I am sorry I was busy. I am documenting LSD_SLAM and make it easier to understand and use [LINK](https://github.com/wow2006/lsd_slam). It will take some time then I will make same for DSO too. I hope that will help others.

Best regards,
Ahmed Abdel Aal

s3r637, 2017-06-29:
Hi @wow2006 

don't worry! :)

I have also some updates in terms of frame poses. Here the skeleton of my `PoseOutputWrapper`.

```
void printPose(Sophus::SE3d const & camToWorld)
{
	auto const m = camToWorld.matrix3x4();
	std::cout  << m(0,0) << " " << m(0,1) << " " << m(0,2) << " " << m(0,3) << " "
			<< m(1,0) << " " << m(1,1) << " " << m(1,2) << " " << m(1,3) << " "
			<< m(2,0) << " " << m(2,1) << " " << m(2,2) << " " << m(2,3) << "\n";
}

virtual void publishKeyframes(std::vector<dso::FrameHessian*> & frames, bool final, dso::CalibHessian*)
{
	if (final) {
		for(dso::FrameHessian* f : frames) {
			std::cout "+ " << frame->incoming_id << " ";
			printPose(f->shell->camToWorld);
		}
	}
}

virtual void publishCamPose(dso::FrameShell* frame, dso::CalibHessian*)
{
	std::cout "- " << frame->incoming_id << " ";
	printPose(frame->camToWorld);
}
```
Best regards

DxsGeorge, 2017-07-10:
What are the arguments for ` dso::FrameShell* frame` and `dso::CalibHessian*` in your main.cpp?

s3r637, 2017-07-19:
Hi @DxsGeorge 

I wrote my own Output3DWrapper (see README) and pushed it into the FullSystem. No further changes!

cheers

DxsGeorge, 2017-07-20:
@s3r637 Thank you, I will check it out!

NikolausDemmel, 2017-12-07:
Closing since the question seems answered / due to lack of activity.

closed 2017-12-07 by NikolausDemmel

farhanmubasher, 2018-04-23:
Hi @s3r637 
Can you please help with this:
I have got poses in 3x4 matrix using camToWorld. But i don't know how to use this matrix, how can i use this matrix to measure distance which is covered by camera ?

DRAhmadFaraz, 2019-10-15:
@s3r637 

I gave 3000+ images as input to DSO algo and got only 500 poses as output in text file. I don't know why this algo missed the 2500 image's poses so I am not sure these 500 poses are of which images !
**how to get the poses for every corresponding images.?**

same problem on TUM dataset and custom dataset.

Thanx in advance.

Regards

Shivani1796, 2019-12-23:
@DRAhmadFaraz 
hey, i am also working on DSO and i want to know that how you created the pose text file because i a unable to create for my own dataset can you explain the way how you get it?
thanks