JakobEngel/dso #55: Extracting point cloud

ginko3, 2017-05-06:
Hi, I've been trying to extract the final point clound from dso, to later use it in a visualizer (meshlab). I looked around `publishKeyframes()` but it only seems to be talking about pointHessians, and not the actual point cloud.
The results I get from extracting the content of the buffer in `refreshPC()` doesn't represent the final result either, it may be because I'm not sure how to use the `camToWorld` projection matrix.
Has anyone managed to get pass this issue ?

s3r637, 2017-05-08:
Hi @ginko3,

it should work like following:
```
 ...
for(auto * f : frames) {
    ...
    auto const & m =  f->shell->camToWorld.matrix3x4();
    // use only marginalized points
    auto const & points = f->pointHessiansMarginalized;
    ...
    for (auto const * p : points) {
        auto const x = ...
        auto const y = ...
        auto const z = ...;
        ... camPoint(x, y, z, 1.f);
        ... worldPoint = m * camPoint;
    }
}
...
```
 [Here](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/IOWrapper/Pangolin/KeyFrameDisplay.cpp#L242) is shown how you can calculate (`camPoint`) the 3D point in camera frame.

Hope this helps

ginko3, 2017-05-08:
Hi,
thanks for your answer, I'm now able to export a real point cloud !
However there is only a few number of point getting exported compared to what Pangolin displays, and sometimes the 3D model isn't even recognizable. I tried exporting other Hessian points, without much more results.
Is there still something I am missing ?

s3r637, 2017-05-09:
Hi @ginko3,

in Pangolin a point is represented by up to eight noisy/random points ([see here](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/IOWrapper/Pangolin/KeyFrameDisplay.cpp#L235)). Marginalize points are right. If you have a `if (final)` block around your code, other `PointHessian` vectors should be empty. I believe you shouldn't use `patternP` for `x` and `y` calculation.

```
int dx = patternP[idx][0]; // is not needed
...
auto const x = (p->u * fxi + cxi) * depth; // not ((p->u + dx) * fxi + cxi) * depth;
....
```

Change this [line](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp#L127) as follow:
```
pangolin::Var<int> settings_pointCloudMode("ui.PC_mode",0,0,4,false);
```
and this [line](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp#L147)
```
pangolin::Var<int> settings_sparsity("ui.sparsity",20,1,20,false);
```

Now consider only the blue points. It should looks more as yours extracted cloud.

cheers

ginko3, 2017-05-12:
Hi,
ok I get it now, i was just worried some points were missing.
Anyway, thanks again for your answer, everything is running smoothly now !

closed 2017-05-12 by ginko3

Neoplanetz, 2017-12-18:
@ginko3 Hi. I saw your issue in here. I am also want to extract final Point cloud to visualize on the RViz and use it for another purpose. Can you solve this problem? And Could you explain how to extract real final point cloud like Pangolin view or could you share code for extracting final point cloud? 

I'm waiting for your kind answer and it will be helpful for many people who want to use DSO to extract point cloud(a lot of people inquire it many times in here)

ginko3, 2017-12-18:
Hi @Neoplanetz, sorry not to have provided a clear answer before closing this thread.

Everything is pretty much here, in order to extract the 3D points, you need to modify the `publishKeyframes` function, in the `SampleOutputWrapper.h` file. You can follow @s3r637 code in his first comment, and get the x, y, z, and depth from each frame, as in [here](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/IOWrapper/Pangolin/KeyFrameDisplay.cpp#L242). It should look something like that : 

```
float depth = 1.0f / p->idepth;
auto const x = (p->u * fxi + cxi) * depth;
auto const y = (p->v * fyi + cyi) * depth;
auto const z = depth * (1 + 2*fxi);
```

`fxi`, `cxi`, etc... being defined at the beginning of the function.

Each frame contains the points' coordinates x y z, with respect to the current camera position. So in order to get the coordinates in Cartesian, you need to multiply them by the cam2world matrix, for example : 
```
Eigen::Vector4d camPoint(x, y, z, 1.f);
Eigen::Vector3d worldPoint = m * camPoint;
```

I hope that helped, don't hesitate if you have further questions.

shikailei, 2017-12-26:
@ginko3 Hi,could you share your code for how to save the point cloud?Thank you  very much

shikailei, 2017-12-26:
@Neoplanetz can you save the point cloud now ?

ginko3, 2018-01-09:
Hi, these are the modification i have made in order to save the 3D points to a .ply file. In order to open it with software like meshlab or blender, you will have to add a header.

https://github.com/ginko3/dso/commit/01a31523ef97ea1956eb753b9cf999ad15dc52f9

Neoplanetz, 2018-01-09:
@ginko3 Thanks @ginko3 . Actually, I made it for your advice. That's all about making PLY file for point cloud. 
@shikailei  In my case, I made a button about "Save Point Cloud" in DSO Panglin.
When I have a spare time, I'll share it on Github.

[image]

yikuangteh, 2018-01-12:
@ginko3 I have attempted to plot the points from the generated .ply file in MeshLab, but it does not seem to be working. I have put in the header with the respective elements needed, but it is not plotting exactly like on Pangolin (Everything seems cluttered at one area).
Is there a face value from the DSO that I'm supposed to figure out?

ginko3, 2018-01-14:
@yikuangteh unfortunately I have bumped into this issue many times myself, and it can be the result of various problems, so I cannot give you a proper solution without more information. You could try a bit of easy debugging, for example checking the [camera matrix](https://docs.opencv.org/2.4/doc/tutorials/calib3d/camera_calibration/camera_calibration.html) is correctly defined, the points are streamed until the end of the execution, switching to another 3D viewer (blender for example)...

yikuangteh, 2018-01-15:
@ginko3 Thank you for the suggestions! It seems MeshLab is extremely bad at scaling. I used a TUM Dataset to plot the points, however it plotted the points found at the sky as well. Thus, there are coordinates plotted beyond the usual range of values (Eg. X-coordinate: 2058.83), making the map look extremely small if not scaled properly. Using Blender fixed the issue.

ginko3, 2018-01-15:
@yikuangteh great, faced with the same issue I had to write a bit of code to remove these points, but I did not included it as it was more of an ugly fix than a real solution.

mrwtakev, 2018-03-15:
@Neoplanetz Hello,I saw your answer in here.And in your anwser you say you "made a button about "Save Point Cloud" in DSO Panglin" . I need "save point cloud" so much ,can you share your project?Thank you.

mrwtakev, 2018-03-15:
@ginko3 Hello,I need to use dso generate point cloud and I saw your anwser.And then I use your code and run the project but No point cloud file generated.The result file is the same as the original. Can you tell me what I should do to get a point cloud?Thank you.

Neoplanetz, 2018-03-15:
@mrwtakev  I made a github repository for DSO with saving Point Cloud data. You can refer to that now.

mrwtakev, 2018-03-15:
@Neoplanetz Thank you very much ! And the program crashes when I run the program.In addition,Also, when I click savePointCloud, the visual interface will stop working.Have you encountered these problems?Thank  you.

Neoplanetz, 2018-03-15:
@mrwtakev  I just tested it and nothing is problem.
[image]

This is my sequence for installing and executing dso_dataset.

It assume that you already installed dso before. So I don't install defendency lib like Pangolin, ziplib ETC.

1. Git clone from my repository.
`git clone https://github.com/Neoplanetz/dso_with_saving_pcl.git`

2. Rename 'dso_with_saving_pcl' folder name to 'dso'

3. Build it
`cd dso
mkdir build
cd build
cmake ..
make -j`

4. Download dataset from https://vision.in.tum.de/mono-dataset and copy and extract that to /dso/build/bin/

5. Excute (I used 'sequence_10' data set)
`./dso_dataset files=`sequence_10/images.zip calib=sequence_10/camera.txt gamma=sequence_10/pcalib.txt vignette=sequence_10/vignette.png preset=0 mode=0 sampleoutput=1`

And when all of frames are finished, you just push the saving button.

mrwtakev, 2018-03-15:
@Neoplanetz Thank you very much ! I succeeded after adding sampleoutput=1.

chapa17, 2021-11-04:
Hallo @s3r637 

I have successfully published the Marginalized points with ROS. But they seem to be very less in count as you explained. I want to publish all the points that Pangolin displays I went through the code files KeyFrameDisplay and PangolinDSOViewer but could not get enough idea. Can you please help me with this?

I would be glad if someone else can also provide me with some hint.

Thank you.

s3r637, 2021-11-04:
Hello @chapa17,

please see the first sentence of [comment#300107540](https://github.com/JakobEngel/dso/issues/55#issuecomment-300107540). (If this is not enough, please let me know.)

Cheers.

chapa17, 2021-11-04:
Hello @s3r637 ,

Thank you for your quick response. I tried the steps mentioned in the comment but I have very few points as a result which does not make it any better for study. That's why I am trying to extract all the points published in the Pangolin viewer. Since what I currently get is only few floating points although they are the true points for one of the sequences in TUM dataset.

(PS: I am working with RViz)

s3r637, 2021-11-04:
Hi @chapa17,

(I just want to make sure you didn't miss the first sentence: "in Pangolin a point is represented by up to eight noisy/random points".)

use these settings in the Pangolin Viewer to view the "raw" (not bloated and not filtered) point cloud.
[image]
If your published points look the same, you are fine.
 
Cheers.

s3r637, 2021-11-08:
Hey @chapa17,

the point cloud computation looks good to me. Save the point cloud to a file in a format for which there is a visualizer...
Since I don't know how (the) ROS (visualizer) expects the data, I can't help. What is the purpose of the transformation in the tf, how is it used, how it is linked etc... idk.

Cheers.

chapa17, 2021-11-08:
Hey,

Thanks a lot! I think I have figured out the problem. I tried the same solution as you suggest saving it into a pcd file and then viewing on RViz. It looks almost similar to the results on Pangolin Viewer. Hence, I concluded there is some issue when I convert the cloud points to ROS message. 

Best regards

jonathantyl97, 2023-12-12:
> @mrwtakev I just tested it and nothing is problem. [image]
> 
> This is my sequence for installing and executing dso_dataset.
> 
> It assume that you already installed dso before. So I don't install defendency lib like Pangolin, ziplib ETC.
> 
> 1. Git clone from my repository.
>    `git clone https://github.com/Neoplanetz/dso_with_saving_pcl.git`
> 2. Rename 'dso_with_saving_pcl' folder name to 'dso'
> 3. Build it
>    `cd dso`
>    `mkdir build`
>    `cd build`
>    `cmake ..`
>    `make -j`
> 4. Download dataset from https://vision.in.tum.de/mono-dataset and copy and extract that to /dso/build/bin/
> 5. Excute (I used 'sequence_10' data set)
>    `bin/dso_dataset files=bin/sequence_10/images.zip calib=bin/sequence_10/camera.txt gamma=bin/sequence_10/pcalib.txt vignette=bin/sequence_10/vignette.png preset=0 mode=0 sampleoutput=1`
> 
> And when all of frames are finished, you just push the saving button. You can find 'pcl_data.pcd' file on '/dso/build/bin/'

Hi! thanks for the good work. I have tried your version and it works really well. I would like to know that, after I push the saving button, the result.txt only consist of very little amount of point cloud (~120 ids) information? My dataset will give out more than 35000 point clouds when I run DSO.

Thanks for helping. Cheers!