JakobEngel/dso #57: Compilation on Jetson TK1

maheshcn, 2017-05-16:
I am trying to compile DSO on jetson tk1 (armhf) . But sse flags are enabled. So the compilation of Optimaztion code is failed. Is any work around available ?

`[  4%] [  8%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o
Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemOptimize.cpp.o
In file included from /home/ubuntu/autoboat/packs/dso/src/FullSystem/CoarseTracker.h:32:0,
                 from /home/ubuntu/autoboat/packs/dso/src/FullSystem/FullSystem.cpp:47:
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:121:11: error: ‘__m128’ does not name a type
     const __m128 val)
           ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:121:18: error: ISO C++ forbids declaration of ‘val’ with no type [-fpermissive]
     const __m128 val)
                  ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:137:11: error: ‘__m128’ does not name a type
     const __m128 val)
           ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:137:18: error: ISO C++ forbids declaration of ‘val’ with no type [-fpermissive]
     const __m128 val)
                  ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h: In member function ‘void dso::Accumulator11::updateSSE(int)’:
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:123:56: error: ‘_mm_load_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                        ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:123:61: error: ‘_mm_add_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                             ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:123:62: error: ‘_mm_store_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                              ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h: In member function ‘void dso::Accumulator11::updateSSENoShift(int)’:
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:139:56: error: ‘_mm_load_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                        ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:139:61: error: ‘_mm_add_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                             ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:139:62: error: ‘_mm_store_ps’ was not declared in this scope
    _mm_store_ps(SSEData, _mm_add_ps(_mm_load_ps(SSEData),val));
                                                              ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h: In member function ‘void dso::Accumulator11::shiftUp(bool)’:
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:157:59: error: ‘_mm_load_ps’ was not declared in this scope
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                                                           ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:157:83: error: ‘_mm_add_ps’ was not declared in this scope
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                                                                                   ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:157:84: error: ‘_mm_store_ps’ was not declared in this scope
     _mm_store_ps(SSEData1k, _mm_add_ps(_mm_load_ps(SSEData),_mm_load_ps(SSEData1k)));
                                                                                    ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:164:61: error: ‘_mm_load_ps’ was not declared in this scope
     _mm_store_ps(SSEData1m, _mm_add_ps(_mm_load_ps(SSEData1k),_mm_load_ps(SSEData1m)));
                                                             ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:164:85: error: ‘_mm_add_ps’ was not declared in this scope
     _mm_store_ps(SSEData1m, _mm_add_ps(_mm_load_ps(SSEData1k),_mm_load_ps(SSEData1m)));
                                                                                     ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:164:86: error: ‘_mm_store_ps’ was not declared in this scope
     _mm_store_ps(SSEData1m, _mm_add_ps(_mm_load_ps(SSEData1k),_mm_load_ps(SSEData1m)));
                                                                                      ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h: At global scope:
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:283:11: error: ‘__m128’ does not name a type
     const __m128 J0,const __m128 J1,
           ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:283:18: error: ISO C++ forbids declaration of ‘J0’ with no type [-fpermissive]
     const __m128 J0,const __m128 J1,
                  ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:283:27: error: ‘__m128’ does not name a type
     const __m128 J0,const __m128 J1,
                           ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:283:34: error: ISO C++ forbids declaration of ‘J1’ with no type [-fpermissive]
     const __m128 J0,const __m128 J1,
                                  ^
[... 355 lines cut ...]
    _mm_store_ps(pt, _mm_add_ps(_mm_load_ps(pt),_mm_mul_ps(J8w,J8))); pt+=4;
                                                           ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h: In member function ‘void dso::Accumulator9::shiftUp(bool)’:
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:1327:68: error: ‘_mm_load_ps’ was not declared in this scope
      _mm_store_ps(SSEData1k+4*i, _mm_add_ps(_mm_load_ps(SSEData+4*i),_mm_load_ps(SSEData1k+4*i)));
                                                                    ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:1327:96: error: ‘_mm_add_ps’ was not declared in this scope
      _mm_store_ps(SSEData1k+4*i, _mm_add_ps(_mm_load_ps(SSEData+4*i),_mm_load_ps(SSEData1k+4*i)));
                                                                                                ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:1327:97: error: ‘_mm_store_ps’ was not declared in this scope
      _mm_store_ps(SSEData1k+4*i, _mm_add_ps(_mm_load_ps(SSEData+4*i),_mm_load_ps(SSEData1k+4*i)));
                                                                                                 ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:1336:70: error: ‘_mm_load_ps’ was not declared in this scope
      _mm_store_ps(SSEData1m+4*i, _mm_add_ps(_mm_load_ps(SSEData1k+4*i),_mm_load_ps(SSEData1m+4*i)));
                                                                      ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:1336:98: error: ‘_mm_add_ps’ was not declared in this scope
      _mm_store_ps(SSEData1m+4*i, _mm_add_ps(_mm_load_ps(SSEData1k+4*i),_mm_load_ps(SSEData1m+4*i)));
                                                                                                  ^
/home/ubuntu/autoboat/packs/dso/src/OptimizationBackend/MatrixAccumulators.h:1336:99: error: ‘_mm_store_ps’ was not declared in this scope
      _mm_store_ps(SSEData1m+4*i, _mm_add_ps(_mm_load_ps(SSEData1k+4*i),_mm_load_ps(SSEData1m+4*i)));
                                                                                                   ^
make[2]: *** [CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/dso.dir/all] Error 2
make: *** [all] Error 2`

s3r637, 2017-05-19:
Hey @maheshcn,

I'm not totally sure but...
DSO is using some Intel types (like `__m128`) which may aren't compatible with arm processors. Two possible solutions come into my mind:

1. look for cross compilation support, maybe the compile or preprocessors is able to replace them during compilation with the target types
2.do-it-yourself, check for equivalent types and replace them with simple command line tools

Maybe this helps

israelshirk, 2017-05-22:
This is a repeat of #31 - here's a site with some info on converting the Intel SSE instructions to ARM NEON instructions: http://codesuppository.blogspot.com/2015/02/sse2neonh-porting-guide-and-header-file.html

I'll likely throw an automated script to do this translation for me in the next day or two, but it's easy enough off the blog post and source.

israelshirk, 2017-05-24:
Easier than expected - just add the header and go.  #59 should take care of it.

maheshcn, 2017-05-25:
checkout the version https://github.com/israelshirk/dso
copy sse2neon to thirdparty folder . 
add -mfpu=neon also in the flags. Install  gcc-multilib g++-multilib  .
It works with jetson tk1 L4T.

Yelen719, 2017-06-05:
May I ask you how did you compile BOOST on Jetson?

JakobEngel, 2017-06-09:
merged the pull request, seems to be alright now :)
thanks to israelshirk!

closed 2017-06-09 by JakobEngel