JakobEngel/dso #60: camera.txt not found (core dumped)

shim94kr, 2017-05-24:
Hello, I built the dso package successfully and downloaded sequence_26 from tum's mono_dataset.. 
But I couldn't figure out why the program can't get the camera.txt as an input.
I have extracted the sequence in dso/build/bin/ folder.

'bin/dso_dataset files=../sequence_26/images calib=../sequence_26/camera.txt gamma=../sequence_26/pcalib.txt vignette=../sequence_26/vignette.png preset=0 mode=0'

loading data from ../sequence_26/images!
loading calibration from ../sequence_26/camera.txt!
loading gammaCalib from ../sequence_26/pcalib.txt!
loading vignette from ../sequence_26/vignette.png!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
PHOTOMETRIC MODE WITH CALIBRATION!
**Reading Calibration from file ../sequence_26/camera.txt ... not found. Cannot operate without calibration, shutting down.**
Segmentation Fault (core dumped)

Can anyone help about this?

DxsGeorge, 2017-05-29:
Can you try to give the absolute path?

shim94kr, 2017-05-30:
I also thought that the file path was wrong rather than the code.
So I solved this issue by moving dataset to main directory of dso and using same command!

bin/dso_dataset files=../sequence_26/images calib=../sequence_26/camera.txt gamma=../sequence_26/pcalib.txt vignette=../sequence_26/vignette.png preset=0 mode=0

Thank you for your consideration.

closed 2017-05-30 by shim94kr