JakobEngel/dso #62: Trying to write a c++ module to run from webcam

DxsGeorge, 2017-05-29:
Anyone has any idea or example how is this possible? I have tried modifying the dso_ros or the dso_main_pangolin but I get eigen errors.

DxsGeorge, 2017-06-06:
For some reason when I initialize FullSystem with 
`FullSystem *fs = new FullSystem;` 
it crashes with Eigen error on CalibHessian.h 
When I just use 
`FullSystem fs;` 
it runs.

JakobEngel, 2017-06-09:
sounds like some alignment issue, which should be prevented by the EIGEN_MAKE_ALIGNED_OPERATOR_NEW in FullSystem.h.

DxsGeorge, 2017-06-09:
The more weird thing is, what I have written now sometimes crashes with EIGEN error and sometimes runs completely. EIGEN must have done something really weird in 3.3 . Anyways thank you for your answer and the code of course!

closed 2017-07-10 by DxsGeorge