JakobEngel/dso #63: inspection robot inside a pipe

rAm1n, 2017-06-07:
Hello,

I am trying to test DSO with a dataset recorded by an inspection robot inside a pipe. It fails to generate any trajectory but, It seems that it's doing not so bad at detecting  some features and points.

Do you have any recommendation for me?

here is a sample from the dataset: 

[image]

Thanks for sharing the code.

chabaudmorgan, 2017-10-23:
The image looks really dark, it might be an issue but I never tried such illuminations conditions... so might as well not be the issue.

BUT, the pole on the image center is not looking good... Is it fixed on the robot? If so, DSO thinks every little thing in the image is the environment... If the pole moves with the camera, it will produce good amounts of errors.

Since you asked for this a long time ago... You may have found your answer already.  But eh!

NikolausDemmel, 2017-12-07:
Closing since the question seems answered / due to lack of activity. Feel free to continue commenting and the issue can be reopend on demand.

closed 2017-12-07 by NikolausDemmel