JakobEngel/dso #64: Error while linking executable

andrespy, 2017-06-08:
Hello I'm having some trouble with the last part of the installation. Here's the error:
```bash
[ 96%] Linking CXX static library lib/libdso.a
[ 96%] Built target dso
[ 96%] Building CXX object CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o
[ 96%] Linking CXX executable bin/dso_dataset
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFWriteEncodedStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libsoxr.so.0: undefined reference to `GOMP_parallel@GOMP_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFGetField@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFNumberOfStrips@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFClose@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFOpen@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFSetField@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
CMakeFiles/dso_dataset.dir/build.make:165: recipe for target 'bin/dso_dataset' failed
make[2]: *** [bin/dso_dataset] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/dso_dataset.dir/all' failed
make[1]: *** [CMakeFiles/dso_dataset.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
Thank you in advance

DxsGeorge, 2017-06-08:
Does /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0 exist?

JakobEngel, 2017-06-09:
one possible reason for this kind of error is that you have multiple versions of openCV installed (e.g., one from ROS, and one as system library), such that there is a mismatch between headers used during compilation, and and binaries used during linking.

andrespy, 2017-06-09:
I'm sorry the problem was that I didn't correctly install the libraries. Everything works fine! Thank you and sorry for the inconveniences

closed 2017-06-09 by andrespy

bonlime, 2017-07-05:
@andrespy  i have the same issue installing dso. What did you do in order to fix it? I already tried removing and reinstalling all packages from Installation, it didnt help

tjacobs, 2017-09-05:
Same problem, was it the libtiff library that needed reinstalling?

nunuwin, 2017-11-30:
I have the same error, I installed opencv3.30 and vision_opencv. Do you means from these, should be only one? or opencv 3.30 and opencv3.10?

nunuwin, 2017-11-30:
How should I do?

nunuwin, 2017-11-30:
After I uninstalled and redo again but error is the same.... Like that

' Found LIBZIP: /usr/local/lib/libzip.so  
--- found PANGOLIN, compiling dso_pangolin library.
--- found OpenCV, compiling dso_opencv library.
--- found ziplib (1.1.1), compiling with zip capability.
--- compiling dso_dataset.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nu/dso/build
nu@nu-VirtualBox:~/dso/build$ make
Scanning dependencies of target dso
[  4%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o
[  8%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemOptimize.cpp.o
[ 12%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemOptPoint.cpp.o
[ 16%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemDebugStuff.cpp.o
[ 20%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemMarginalize.cpp.o
[ 24%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/Residuals.cpp.o
[ 28%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/CoarseTracker.cpp.o
[ 32%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/CoarseInitializer.cpp.o
[ 36%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/ImmaturePoint.cpp.o
[ 40%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/HessianBlocks.cpp.o
[ 44%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/PixelSelector2.cpp.o
[ 48%] Building CXX object CMakeFiles/dso.dir/src/OptimizationBackend/EnergyFunctional.cpp.o
[ 52%] Building CXX object CMakeFiles/dso.dir/src/OptimizationBackend/AccumulatedTopHessian.cpp.o
[ 56%] Building CXX object CMakeFiles/dso.dir/src/OptimizationBackend/AccumulatedSCHessian.cpp.o
[ 60%] Building CXX object CMakeFiles/dso.dir/src/OptimizationBackend/EnergyFunctionalStructs.cpp.o
[ 64%] Building CXX object CMakeFiles/dso.dir/src/util/settings.cpp.o
[ 68%] Building CXX object CMakeFiles/dso.dir/src/util/Undistort.cpp.o
[ 72%] Building CXX object CMakeFiles/dso.dir/src/util/globalCalib.cpp.o
[ 76%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/OpenCV/ImageDisplay_OpenCV.cpp.o
[ 80%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp.o
[ 84%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/Pangolin/KeyFrameDisplay.cpp.o
[ 88%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp.o
[ 92%] Linking CXX static library lib/libdso.a
[ 92%] Built target dso
Scanning dependencies of target dso_dataset
[ 96%] Building CXX object CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o
make[2]: *** No rule to make target '/home/nu/Pangolin/build/src/libpangolin.so', needed by 'bin/dso_dataset'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/dso_dataset.dir/all' failed
make[1]: *** [CMakeFiles/dso_dataset.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2 '

How should I do?

NikolausDemmel, 2017-11-30:
@nunuwin, that seems like a different issue than OP's. You should probably open a new issue. I you do, please add some more info about your system, compiler, and output logs for cmake and make for both pangolin and dso.

Looks like somehow the pangolin library is not found. Did you compile it properly? Does libpangolin.so exist somewhere?

nunuwin, 2017-12-01:
Yes, cmake is like that...

nunuwin, 2017-12-01:
Now it is ok my error..
I used by that,
'git clone https://github.com/stevenlovegrove/Pangolin.git
cd Pangolin
mkdir build
cd build
cmake -DCPP11_NO_BOOST=1 ..
make -j1
cd ~/dso
.
.
.
make
Now result is...
nu@nu-VirtualBox:~/dso/build$ make
[ 92%] Built target dso
[ 96%] Linking CXX executable bin/dso_dataset
[100%] Built target dso_dataset

NikolausDemmel, 2017-12-01:
I'm not sure what your issue was but I'm glad it's resolved.

zerohuwan, 2023-01-08:
Hello I'm having some trouble with the last part of the installation. Here's the error:
```
~/nikdso/src/dso/build$ make 
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   thread
--   chrono
--   date_time
--   atomic
--- found PANGOLIN, compiling dso_pangolin library.
--- found OpenCV, compiling dso_opencv library.
--- found ziplib (1.1.1), compiling with zip capability.
--- compiling dso_dataset.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zhangfangfei/nikdso/src/dso/build
[ 92%] Built target dso
[ 96%] Linking CXX executable bin/dso_dataset
CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o: In function `main::{lambda()#1}::operator()() const':
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:256: undefined reference to `zip_fopen'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:257: undefined reference to `zip_fread'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:264: undefined reference to `zip_fopen'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:265: undefined reference to `zip_fread'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:256: undefined reference to `zip_fopen'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:257: undefined reference to `zip_fread'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:264: undefined reference to `zip_fopen'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:265: undefined reference to `zip_fread'
CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o: In function `ImageFolderReader::~ImageFolderReader()':
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:170: undefined reference to `zip_close'
CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o: In function `ImageFolderReader::ImageFolderReader(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:125: undefined reference to `zip_open'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:133: undefined reference to `zip_get_num_entries'
/home/zhangfangfei/nikdso/src/dso/src/util/DatasetReader.h:136: undefined reference to `zip_get_name'
collect2: error: ld returned 1 exit status
CMakeFiles/dso_dataset.dir/build.make:152: recipe for target 'bin/dso_dataset' failed
make[2]: *** [bin/dso_dataset] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/dso_dataset.dir/all' failed
make[1]: *** [CMakeFiles/dso_dataset.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```