JakobEngel/dso #68: Coding standards

israelshirk, 2017-06-23:
### Why?
I'm prepping to pull some things apart and wanted to apply some sort of coding standard before doing so.  There are a bunch floating around, so I figured I'd just grab a few and see if you'd like to merge one in so I can keep doing PR's from my codebase.

### Here are the options I came up with:
* LLVM, 4 spaces: https://github.com/israelshirk/dso/tree/format/LLVM-4
* LLVM, 8 spaces: https://github.com/israelshirk/dso/tree/format/LLVM-8
* Google, 4 spaces: https://github.com/israelshirk/dso/tree/format/Google-4
* Google, 8 spaces: https://github.com/israelshirk/dso/tree/format/Google-8
* Chromium, 4 spaces: https://github.com/israelshirk/dso/tree/format/Chromium-4
* Chromium, 8 spaces: https://github.com/israelshirk/dso/tree/format/Chromium-8
* Mozilla, 4 spaces: https://github.com/israelshirk/dso/tree/format/Mozilla-4
* Mozilla, 8 spaces: https://github.com/israelshirk/dso/tree/format/Mozilla-8
* Webkit, 4 spaces: https://github.com/israelshirk/dso/tree/format/Webkit-4
* Webkit, 8 spaces: https://github.com/israelshirk/dso/tree/format/Webkit-8

### The script used for generating these was:
```bash
styles="LLVM Google Chromium Mozilla Webkit"
widths="4 8"

for style in $styles; do
	for width in $widths; do
		git checkout jakob/master
		git checkout -b format/$style-$width
		find src -name '*.h' -o -name '*.hpp' -o -name '*.cpp' -exec clang-format -i -style "{BasedOnStyle: $style, IndentWidth: $width}" {} \;
		git add src
		git commit -m "Formatted with style $style, width $width"
		git push public
	done
done

for style in $styles; do
	for width in $widths; do
		echo "* $style, $width spaces: https://github.com/israelshirk/dso/tree/format/$style-$width"
	done
done
```

###  Are there any that you prefer?

Thanks!  Israel

chabaudmorgan, 2017-07-26:
That would be good to have some consistency with indentation!

However, doing this now would create merging issues for all people already using the code. I wonder if change in coding standards at a late stage of a project is really needed. Plus, right now, it is only visual coding standards so it does not affect the code. Thus the effort needed to use it (check your pull + merge every forked repos) might not be worth it.

I think you just have to close your eyes when you see bad indentations and keep functionality/patch pull requests well indented.

By the way, why no hard-tabs? I love these as every one is able to set their prefered length in their favourite editor and everything is automatically aligned!

israelshirk, 2017-07-28:
@chabaudmorgan At the time of this issue's creation there were near zero branches ahead of the current - and resolving conflicts with any diverged branches could simply be done by applying the chosen transformation.

Tabs added here (I'm a fan but just hadn't gotten to the docs for it yet!)
* LLVM, tabs: https://github.com/israelshirk/dso/tree/format/LLVM-tabs
* Google, tabs: https://github.com/israelshirk/dso/tree/format/Google-tabs
* Chromium, tabs: https://github.com/israelshirk/dso/tree/format/Chromium-tabs
* Mozilla, tabs: https://github.com/israelshirk/dso/tree/format/Mozilla-tabs
* Webkit, tabs: https://github.com/israelshirk/dso/tree/format/Webkit-tabs

israelshirk, 2017-07-28:
Aaaand the script for generating the above:
```bash
styles="LLVM Google Chromium Mozilla Webkit"
widths="8"

for style in $styles; do
	for width in $widths; do
		git checkout jakob/master
		git checkout -b format/$style-tabs
		find src -name '*.h' -o -name '*.hpp' -o -name '*.cpp' -exec clang-format -i -style "{BasedOnStyle: $style, IndentWidth: $width, UseTab: Always}" {} \;
		git add src
		git commit -m "Formatted with style $style, tabs"
		git push public
	done
done

for style in $styles; do
	for width in $widths; do
		echo "* $style, tabs: https://github.com/israelshirk/dso/tree/format/$style-tabs"
	done
done
```

chabaudmorgan, 2017-07-31:
Well, I agree there is not much people having forked this repository but not near zero still... You actually opened this issue in June, and if you take a look at [https://github.com/JakobEngel/dso/network](url), the master has been updated since February... but mainly fixing bugs (or adding ARM compilation). I would say 15-20 people have forked it and worked with it (commits).

Regarding your proposal, I would say "Webkit, tabs" is the best, mainly because it does not wrap automatically the code... And complex or hard to understand code blocks wrapped automatically (without meaning (ex: middle of an equation)) is not doing so good.

NikolausDemmel, 2017-12-07:
While the effort is appreciated @israelshirk, I don't think this will happen for now (@chabaudmorgan gave some valid points). Maybe in a possible future iteration this can be taken into consideration. Closing this for now. Feel free to continue commenting, and we can reopen if needed.

closed 2017-12-07 by NikolausDemmel