JakobEngel/dso #69: How to make dso more "system-lightweight"

DxsGeorge, 2017-06-29:
I am trying to run dso on a raspberry with a live camera. Although I manged to make it run on a laptop, I guess the initialization gets really heavy for the raspberry and I either segfault or straight up freeze. Is there any way to make it more lightweight?

s3r637, 2017-06-29:
Hi @DxsGeorge 

did you check the `preset=2` parameter (see README, 3.2 Commandline Options).
If it doesn't work I recommend you to record a video (image sequences) and try it first on your laptop, if your laptop is able to handle the video then RPi should also handle it.

cheers

DxsGeorge, 2017-06-30:
@s3r637 Thanks for your tips. I will try and give feedback

DxsGeorge, 2017-06-30:
I ran the dso_dataset with euroc images, it's running but it is slow. I'm wondering what can I do to trade performance for accuracy.

s3r637, 2017-07-04:
Try to turn down the following variables [#L131](https://github.com/JakobEngel/dso/blob/a9915821826609ecb87afd04d916fdd8bf224fb4/src/main_dso_pangolin.cpp#L131)
```
setting_desiredImmatureDensity = 600;
setting_desiredPointDensity = 800;
setting_minFrames = 4;
setting_maxFrames = 6;
setting_maxOptIterations=4;
setting_minOptIterations=1;

benchmarkSetting_width = 424;
benchmarkSetting_height = 320;
```
e.g.
```
dso::setting_desiredImmatureDensity = 500;
dso::setting_desiredPointDensity = 650;
dso::setting_minFrames = 2;
dso::setting_maxFrames = 3;
dso::setting_maxOptIterations = 2;
dso::setting_minOptIterations = 1;
// my original resolution: 1920x1200
// maybe the aspect ratio should be the same 
dso::benchmarkSetting_width = 408;
dso::benchmarkSetting_height = 256;
```
I think this is the simplest attackpoint. You could also try to reduce the fps (pass only every second (or third, etc.) image).

cheers

DxsGeorge, 2017-07-06:
Thanks, will try these settings as well!

NikolausDemmel, 2017-12-07:
Closing since the question seems answered / due to lack of activity. Feel free to continue commenting and the issue can be reopend on demand.

closed 2017-12-07 by NikolausDemmel