JakobEngel/dso #70: How to run the code on the kitti dataset

OfirBeckerHome, 2017-06-29:
Hello,
I am trying to run the code as is on the kitti dataset. I'm not sure how to adjust the configuration or the parameters to make it work, as the required calib/gamma/vignette don't exist in that format and the images are undistorted and rectified. Is there a written configuration for the Kitti dataset? or can you explain what needs to be done in order to run such images on the system?

s3r637, 2017-06-29:
Hi @OfirBeckerHome 

see https://github.com/JakobEngel/dso/issues/51#issuecomment-300748656

cheers

NikolausDemmel, 2017-12-07:
Closing since the question seems answered / due to lack of activity. Feel free to continue commenting and the issue can be reopend on demand.

closed 2017-12-07 by NikolausDemmel