JakobEngel/dso #71: How to define values in Calibration File for FOV camera model

atuldiv, 2017-07-03:
Hi,
I am using DSO package. In DSO there is a camera.txt file in that the format is (Calibration File for FOV camera model)  given below :
FOV fx fy cx cy omega
in_width in_height
"crop" / "full" / "fx fy cx cy 0"
out_width out_height

Also I saw in mono_dataset  there is example of camera.txt file they are using as :
0.349153000000000	0.436593000000000	0.493140000000000	0.499021000000000	0.933271000000000
1280 1024
0.4 0.53 0.5 0.5 0
640 480

Now my Camera Calibrated data is :
width : 1280
height : 960
camera matrix : [ 601.571920 0.000000 615.918604 0.000000 582.719805 363.810622 0.000000           0.000000 1.000000 ]
Here fx :  601.571920 ,  fy : 582.719805,     cx : 615.918604, cy : 363.810622,
distortion : [ -0.272383 0.046325 0.001219 0.004647 0.000000]
rectification : [1.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 1.0 ]
projection : [412.205078  0.0  616.977482  0.0  0.0  440.144897  316.117456  0.0  0.0  0.0  1.0  0.0 ]

My question is How to modify camera.txt file using above my camera calibration information because I am confused in fx , fy, cx and cy. Also I don't understand What is  out_width, out_height and in_width, in_height  in camera.txt file. Please give some solution. Thanks in advanced.

NikolausDemmel, 2017-07-05:
I think this is exmplained in the Readme (https://github.com/JakobEngel/dso). See section 3.1.

It looks like you might have a radial-tangential distortion model, so use the "RadTan" and ignore the 5th distortion parameter (it is 0.0 anyway).

The readme also explains that the fx, fy, cx, cy are relative to the total witdth/height, so you need to divide your values and you need to check if you should apply the 0.5 offset or not, depending on your camera.

out_width / out_height is related to the info in "Rectification modes". It simply allows you to resize the image during rectification.

atuldiv, 2017-07-06:
Thanks for suggestion

chabaudmorgan, 2017-07-31:
Why did you reopened this issue?

If you do as @NikolausDemmel said, you should have:
######## CameraFile #########
RadTan 601.57 582.72 615.92 363.81 -0.272383 0.046325 0.001219 0.004647
1280 960
none
1280 960
######## CameraFile #########

If you were asking why mono_dataset example (camera.txt) starts with 0.xxxx values, that is because it uses the convention:
fx/width fy/height cx/width cy/height omega

Note: As mentioned in Readme:
> (note: for backwards-compatibility, "Pinhole", "FOV" and "RadTan" can be omitted)

Hope it removes all doubt...

NikolausDemmel, 2017-08-02:
Thanks @chabaudmorgan, I'll close the issue.

Feel free to continue the discussion. We can re-open the issue if needed.

closed 2017-08-02 by NikolausDemmel

wodemaa5, 2018-12-08:
what the meaning of omega ?  field of view?

fangxu622, 2019-11-07:
@NikolausDemmel @chabaudmorgan  accord the calibration file .

```
calibration_time: "2019年10月09日 星期三 09时56分00秒"
nr_of_frames: 114
image_width: 1920
image_height: 1440
board_width: 9
board_height: 6
square_size: 28.
fix_aspect_ratio: 0.
# flags: +fix_skew +recompute_extrinsic
flags: 522
fisheye_model: 1
camera_matrix: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [ 9.3359114467221752e+02, 0., 9.5950000000000000e+02, 0.,
       9.3996822344250461e+02, 7.1950000000000000e+02, 0., 0., 1. ]
distortion_coefficients: !!opencv-matrix
   rows: 4
   cols: 1
   dt: d
   data: [ 1.3571629546076457e-02, 1.7799816234275553e-01,
       -2.9965104051643743e-01, 1.8437185124189442e-01 ]
avg_reprojection_error: 3.1639901210517574e-01
```

######## CameraFile #########
RadTan 933.591144672	939.968223442	959.50	719.50	1.3571629546076457e-02	1.7799816234275553e-01	-2.9965104051643743e-01	1.8437185124189442e-01
1920 1440
none
1920 1440
######## CameraFile #########

but get the problem like this when running dso, the image distortion

[image]

NikolausDemmel, 2019-11-07:
Looks like you use the opencv fisheye model. This is called equidistant or kanalla brandt in DSO.

fangxu622, 2019-11-10:
> Looks like you use the opencv fisheye model. This is called equidistant or kanalla brandt in DSO.

Yes

NikolausDemmel, 2019-11-10:
So you need to change the camera.txt Accordingly. Check readme or code which is the exact name for that model.

fangxu622, 2019-11-10:
> So you need to change the camera.txt Accordingly. Check readme or code which is the exact name for that model.

Yes，thanks very much. Absolutely understand. and I have to do that.  please check #99

StevePhan101, 2020-08-03:
Hi, when I set the output pars as "none" or "crop" or "full", it reports the error: ```Out: Failed to Read Output pars... not rectifying.```.
Does anyone know the issue? Thank you

qian2020, 2022-04-24:
> Hi, when I set the output pars as "none" or "crop" or "full", it reports the error: `Out: Failed to Read Output pars... not rectifying.`. Does anyone know the issue? Thank you

hello, I also met this problem, did you resolved this problem?