JakobEngel/dso #73: How to setup input files for ICL-NUIM dataset?

youngguncho, 2017-07-06:
Hi. I'm trying to use DSO on ICL-NUIM datasets.
As in the photometric calibration paper, I run DSO without pcalib and vignette files, such as

$ build/bin/dso_dataset calib=calib/calib.txt files=data/original.zip preset=0 mode=2

I first define the calib file as (which from ground truth calib data) 
481.20 480.0 319.50 239.50 0
640 480
none
640 480

But I got assertion error as
===================================================
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
START PANGOLIN!
OpenCV Error: Assertion failed (!buf.empty() && buf.isContinuous()) in imdecode_, file /home/yg/svn/irap/third-party/opencv-3.1.0/opencv/modules/imgcodecs/src/loadsave.cpp, line 490
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/yg/svn/irap/third-party/opencv-3.1.0/opencv/modules/imgcodecs/src/loadsave.cpp:490: error: (-215) !buf.empty() && buf.isContinuous() in function imdecode_
===================================================

Also, I tried relative calib file as
0.751875 1 0.5 0.5 0
640 480
none (or crop)
640 480

But it shows same error also. Does this error occurs by wrong calibration data or do I need pcalib and vignette information for ICL-NUIM also?

NikolausDemmel, 2017-07-06:
```
Pinhole 0.751875 1.00 0.5 0.5 0
640 480
none
640 480
```

works fine for me. If you download the dataset as is, you have to make sure DSO is reading them in the correct order. By default, DSO reads them in some order

```
1.png
10.png
100.png
```

or similar. So either rename them to the expected

```
00001.png
00002.png
...
```

or adapt the sorting of files in the code.

In any case I am not sure where the error message comes from. Also, I am not reading them from a zip file, but from a directory with files, so I'm not sure if it might be different there.

NikolausDemmel, 2017-07-12:
P.S.: Also reading from a zip file (with adapted sort order) works fine for me, so this might be a different issue. Can you run successfully with a different dataset, e.g. https://vision.in.tum.de/data/datasets/mono-dataset ?

youngguncho, 2017-07-13:
Thank you for the reply.
As your first comment, I renamed all files with zero padded names, and it worked find!!

NikolausDemmel, 2017-07-24:
@youngguncho, so maybe you can close this issue?

youngguncho, 2017-07-24:
Sorry, I forgot to close the issue.
Thank you for the help!

closed 2017-07-24 by youngguncho