JakobEngel/dso #74: Pthread error on Ubuntu

alejandronespereira, 2017-07-14:
First of all, my system specifications:

```
$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ uname -or
4.8.0-58-generic GNU/Linux
$ lsb_release -irc
Distributor ID: Ubuntu
Release:        16.04
Codename:       xenial

$ cmake --version
cmake version 3.5.1

```

When I try to build, I get this error:

```
cmake ..
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   chrono
--   date_time
--   atomic
CMake Error at /home/alejandro/dev/Pangolin/build/src/PangolinConfig.cmake:8 (include):
  include could not find load file:

    /home/alejandro/dev/Pangolin/build/src/PangolinTargets.cmake
Call Stack (most recent call first):
  CMakeLists.txt:21 (find_package)

--- found PANGOLIN, compiling dso_pangolin library.
--- found OpenCV, compiling dso_opencv library.
--- found ziplib (1.1.1), compiling with zip capability.
--- compiling dso_dataset.
-- Configuring incomplete, errors occurred!
See also "/home/alejandro/dev/dso/build/CMakeFiles/CMakeOutput.log".
See also "/home/alejandro/dev/dso/build/CMakeFiles/CMakeError.log".

```
The Output and Error files are these

[CMakeError.log.txt](https://github.com/JakobEngel/dso/files/1147971/CMakeError.log.txt)
[CMakeOutput.log.txt](https://github.com/JakobEngel/dso/files/1147972/CMakeOutput.log.txt)

NikolausDemmel, 2017-07-14:
It looks like you might have trouble finding Pangolin. Have you compiled Pangolin successfully? Does the file `/home/alejandro/dev/Pangolin/build/src/PangolinTargets.cmake` exist?

alejandronespereira, 2017-07-14:
I compiled it with no errors, but there is not a PangolinTargets.cmake... 

I'll go with that, thanks!

NikolausDemmel, 2017-07-14:
> I compiled it with no errors, but there is not a PangolinTargets.cmake...

Yeah, it should be there. Do you have the latest version of Pangolin? Maybe try compiling again from scratch...

NikolausDemmel, 2017-07-14:
PS: I believe the pthreads releated error you see in the logs is not an actual error and not related to your issue.

alejandronespereira, 2017-07-14:
Yep, it worked. I compiled Pangolin without ffmpeg support, and at some point it messed up and didn't create the PangolinTargets.cmake file. 

I made a new clean Pangolin build and now DSO built flawlessly. 

Regarding the pthreads, there is a CMake bug associated with that, but it has been solved for my version. Guess it just popped for some weird reason. 

Thanks anyway!

closed 2017-07-14 by alejandronespereira

chapa17, 2021-09-21:
I am getting similar error while installing pangolin for DSO. I downloaded the latest Pangolin files using wget. I tried multiple times making a clean build but still I get the same error.

Can you help me solve it?

Thank you in advance!