JakobEngel/dso #76: Run with my own dataset issue ???

trinamntn08, 2017-07-19:
First, i succeed to run DSO SLAM with Examples. 
So now, i want to run DSO SLAM with my own dataset. I stored my set of images in a folder. But when i run command, i got the error as images below : 

[image]

Can you help me to resolve this issue.

Thanks in advance,

NikolausDemmel, 2017-07-19:
You should start it in a debugger and check where exactly it crashes. Google is your friend for that.

chabaudmorgan, 2017-07-20:
Before starting to debug, I would surely check the issue mentioned on the following line "PhotometricUndistorter: Invalid vignette image size! got 1280 x 1024, expected 1920 x 1080"

trinamntn08, 2017-07-20:
@chabaudmorgan  because my own images is on format 1920x1080 , so vignette image size isn't compatible with my dataset, but i dont't think it's my main problem because i just run on mode=2 with no photometric calibrated.

trinamntn08, 2017-07-20:
@NikolausDemmel ; sorry for my stupid question  , how can i run DSO SLAM on mode Debug . 
I lance gdb bin/dso_dataset files=./sequence_01 ...........
But it didn't work

NikolausDemmel, 2017-07-20:
If you don't want to use vignette, you should be able to not pass the parameter instead of passing a fake one. So I guess you are using simlated data, since you use mode=2.

Passing arguments is either done with `--args` or in the `run` command of gdb. 

http://lmgtfy.com/?q=gdb+tutorial

chabaudmorgan, 2017-07-20:
Also make sure you compile with debug information ( In CMakeLists: "set(BUILD_TYPE Debug)") instead of release.

And I agree with @NikolausDemmel , you should avoid wrong parameters when not necessary.

trinamntn08, 2017-07-20:
I think my issue is as below: 

[image]
[image]
I don't really understand the code in this file; Can you explain it , pls ?

chabaudmorgan, 2017-07-20:
Could you try without changing output resolution in the calibration file?

trinamntn08, 2017-07-20:
yes, but i got the same error

trinamntn08, 2017-07-20:
yeah, it solved. Thanks a lot for your patient help :D

chabaudmorgan, 2017-07-20:
What was the issue then?

Don't forget to close the issue afterwards! =)

trinamntn08, 2017-07-20:
Just don't change the output resolution, i keep it as input resolution 1920x1080. It worked !

closed 2017-07-20 by trinamntn08