JakobEngel/dso #77: Generating point clouds from known camera poses

JohnZ1288, 2017-07-19:
I'd like to use some of the concepts from DSO in my own project (generating point clouds from known camera poses), but I'm getting a bit lost in the code. I've read through and played around with most of it, and I've read through the paper as well as all issues here and on dso_ros.

Could someone (perhaps @JakobEngel ) point me to the portions of the code that encompass:

1. Point selection
2. Tracking points/residuals between frames
3. Depth estimation (this would be especially useful)

Thanks!

chabaudmorgan, 2017-07-26:
In order to generate point cloud from known camera poses, I am not sure DSO is what you want (at least, not as it is) because DSO estimate the point clouds **and** the camera poses. But both are intrinsically related.

Considering you know the camera poses, it is all about matching points between images. With this matching and the camera positions (and camera (intrinsic) parameters), you have the point cloud. I could give some references but if you just want "off-the-shelf" codes, you should be able to find something with OpenCV (The following also estimates position: [http://docs.opencv.org/trunk/d4/d18/tutorial_sfm_scene_reconstruction.html](url)). Or you could look for "structure from motion" or "3D reconstruction" and try to find codes asking for camera positions in input.

Nevertheless, the answer to the points you asked for:

1. Point selection: **FullSystem::makeNewTraces** add new immature points (candidate that can be selected for the tracking (and optimisation) when needed) selected by ** PixelSelector::makeMaps**

2. Track points between frames: **FullSystem::traceNewCoarse** ask all immature points on all activated frames to be tracked (only updating their min and max idepth) with the call to **ImmaturePoint::traceOn**

2. (2bis) Computing new residual for tracked points: The following loop in **FullSystem/makeKeyFrame** : [https://github.com/JakobEngel/dso/blob/master/src/FullSystem/FullSystem.cpp#L1070](url)

3. Depth estimation: For immature points, I have mentioned it in (2.). And for tracked points, it is in the optimization process (**FullSystem::optimize**) and here, I recommand to search for 'depth' as the optimization procedure can change the depth "as wished"

But again... This code is not well fitted to do what you want (as it does so much more):
> [generate] point clouds from known camera poses

Hope it helps!

NikolausDemmel, 2017-12-07:
Closing due to inactivity. Feel free to comment and we can reopen if needed.

closed 2017-12-07 by NikolausDemmel

M1234Thomas, 2019-06-13:
Hey @JohnZ1288 , 

I also have a similar interest in generating point clouds from known camera poses. 

Did you figure out any way to do that? 

Even if the algorithm is initialized with ground truth camera poses, the optimization module woild overwrite it right? 

Do you have any idea about it?