JakobEngel/dso #78: How to relocation by saving an offline map

zgq91, 2017-07-21:
Hi:
    how to create an offline map by saving keyframes and relocate it

    Looking forward to your reply.

NikolausDemmel, 2017-07-21:
There is nothing like that in DSO. As is, it is just a keyframe odometry with no concept of a permanent map. But you should be able to extend it using ideas from state-of-the art systems in that area. You have to take care of properly dealing with the marginalized points and keyframes though.

zgq91, 2017-07-25:
@NikolausDemmel  thank you very much for the reply, I need to save mappingLoop  in  the  FrameHessian *h  Variable ?

chabaudmorgan, 2017-07-26:
I am not sure of what you mean by 

> I need to save mappingLoop in the FrameHessian *h Variable ?

But in order to save keyframes to files, you could use the sample output wrapper as an example (IOWrapper/OutputWrapper/SampleOutputWrapper.cpp), and especially the function publishKeyframes which is called when keyframes are marginalized.

You could add some code to write needed information in a file for each marginalized keyframes.

zgq91, 2017-07-27:
@chabaudmorgan Similar to the ORB-SLAM2 relocation function, positioning is done while saving offline maps

chabaudmorgan, 2017-07-27:
Oh I got it all wrong, you don't want to save it in a file!

> I am not sure of what you mean by

>>    I need to save mappingLoop in the FrameHessian *h Variable ?

mappingLoop is a function (not really used if you use default settings) and you want to save it into a FrameHessian ? It doesn't make any sense...

If you meant the concept (the whole mapping loop), it would be camera positions and images only. You could save these in each FrameHessian but I guess it makes more sense to have it in a global way as these FrameHessian are created for each image... 

Then you could check if the SE3 tracker fails, you look for a similar image/position in your offline map.

NikolausDemmel, 2017-09-05:
I'll close this since there are no immediate plans to implement this. Feel free to continue to discuss and we can reopen if needed.

closed 2017-09-05 by NikolausDemmel