JakobEngel/dso #79: cross compilation for android

Edmospheric-pressure, 2017-07-24:
Hi, I have been trying to build a static library for android. I tried to build it using android-cmake using the current cmakelist.txt in the dso folder, but it seemed to have many dependencies issue. I am really not that familiar with building libraries, would you mind to provide some directions or guidance? Many thanks.

chabaudmorgan, 2017-07-26:
I tried once without any success however so I will not be able to help much.

But I think this is not the right place for such a question as it is not directly related to DSO but it is more about cmake and cross-compilation. You should ask in forums more dedicated to this (StackOverflow might be a good one).

Good luck with this!

Please close this issue if you do not mind. You can wait a few days to see if you get a better answer but I would not bet on it.

Edmospheric-pressure, 2017-07-31:
Thank you for the your suggestions. I have successfully cross-compiled a static library for android.  Now, I just have questions about how to read camera pose information from fullsystem, but I will open a new post for that.

closed 2017-07-31 by Edmospheric-pressure