JakobEngel/dso #83: can fullsystem recover from lost?

Edmospheric-pressure, 2017-07-31:
When isLost is true, can fullsystem reinitialize itself? Or should I manually add a condition like "if (isLost) {setting_fullResetRequested = true}?  One more question about the reference frame of the camera pose: is it x(right) y(up) z(pointing away)?

chabaudmorgan, 2017-08-01:
Haven't you tried to make it fail to track? 

Yes it does stop and you can see it in main_dso_pangolin.cpp, after isLost is set to false in FullSystem::addActiveFrame, the following condition is true (L 521) :
`if(fullSystem->isLost)
            {
                    printf("LOST!!\n");
                    break;
}`

This makes it stop but you could put it in front of the previous condition checking for setting_fullResetRequested (`if(fullSystem->initFailed || setting_fullResetRequested)`) and set that variable to false instead of breaking.

As for the reference frame, you could have tested it too. But I have a trajectory going forward/left and another going foward/bottom and the outputs ("result.txt") follow the convention (looking at the image plane (same as [this introduction in OpenCV docs](http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html))):
x(right) y(bottom) z(pointing forward)

Testing is a good way to check things out!

NikolausDemmel, 2017-08-02:
Thanks @chabaudmorgan, I agree with your assessment.

Note that while it should be quite easy to add complete re-initialization after tracking is lost, you will initialize to arbitrary new scale, which could be quite different from the previous scale. One could investigate ways to try to re-align to previous keyframes after re-initialization, but that will be more involved.

I'll close the issue for now. Feel free to continue the discussion. We can re-open the issue if needed.

closed 2017-08-02 by NikolausDemmel