JakobEngel/dso #84: Which part of code indicate the loop closure?

changliu816, 2017-08-02:


NikolausDemmel, 2017-08-02:
Which loop closure are you referring to? Have you read the paper?

changliu816, 2017-08-02:
I have used the program on road test. I drove a circle, and record the images at 60fps. However, when I did the dvo slam, the trajectory is not a close loop. Unlike the lsd_slam code where I can anchor TrackableKeyFrameSearch.cpp as loop closure detection, I failed to find where the dvo code does this job.

NikolausDemmel, 2017-08-02:
DSO does not implement loop closure like LSD-SLAM does, it is a keyframe odometry, not a full SLAM system. Check the paper for more details.

I'll close the issue, but feel free to continue the discussion if you want to and we can re-open as needed.

closed 2017-08-02 by NikolausDemmel