JakobEngel/dso #85: coarseTracker->lastRef in trackNewCoarse points to NULL

Edmospheric-pressure, 2017-08-03:
Hi,

        The dso library ran perfectly with sample video in linux. When I cross-compiled with android and ran with android in real time, I have encountered this error "SIGSEGV (signal SIGSEGV: invalid address (fault address: 0x4))"  after running for a few seconds. The debugger stopped at the line as shown in the attached image. In addition, the debugger also shown that the addresses stored in AllFrameHistory are {NULL,0x1,0x2...}. Has anyone encountered this problem before? 

Best,
Ed 
[image]

chabaudmorgan, 2017-08-04:
I don't know if people tried it on Android but there might be some relation here. Indeed, your sample video (set of images?) worked on linux, and not on Android.

> coarseTracker->lastRef in trackNewCoarse points to NULL

The constructor of CoarseTracker put it to 0 (~ NULL) and until setCoarseTrackingRef is called, it will stay this way.

The first call to trackNewCoarse should happen straight after the system is initialized ([here in addActiveFrame(...)](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/FullSystem/FullSystem.cpp#L861)) At the end of the initialization, deliverTrackedFrame is called, which call makeKeyFrame because needKF is set to true. This function (makeKeyFrame) should call coarseTracker_forNewKF->setCoarseTrackingRef (...) if the system does not consider itself lost ([here](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/FullSystem/FullSystem.cpp#L1128)) When the next image is processed, coarseTracker will be set to coarseTracker_forNewKF ([in addActiveFrame(...)](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/FullSystem/FullSystem.cpp#L853)). Eventually trackNewCoarse will be called ...

So, you can put a "cout" [here](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/FullSystem/FullSystem.cpp#L1137) that will check if the last initialization step call setCoarseTrackingRef. And you can put another one ("cout") [here](https://github.com/JakobEngel/dso/blob/2c30ab44b3804d9e507ba9221dbd9f5637491f12/src/FullSystem/FullSystem.cpp#L856) that will check whether the tracking reference is swapped or not.

However... I guess it is "system-related". So maybe an allocation failed somewhere (lack of memory on the Android device ?) or Pangolin is not fit for Android? I guess you will have to figure that out if no one tried it. =p

I hope the information I gave you will help you find what the issue is. And if you find the issue, please give the community some feedback on it! (You did not on your last one, even if it was not really DSO related, it might help others! )

Thanks

Edmospheric-pressure, 2017-08-05:
Thank you for your help! I think it is most likely the problem of my jni part, since the dso library is working good on those TUM datasets. I disabled Pangolin when I cross-compiled the library, so it should not affect library. Lack of memory might be an issure too, since my device only has 3 gb of ram. I will update more if I find out what the problems are.

chabaudmorgan, 2017-08-05:
Yup!

Well 3Gb of RAM looks good, I can run DSO on a set of 1000 images (res: 512x512) without going over 250Mb of RAM.

NikolausDemmel, 2017-12-07:
Closing since the question seems answered / due to lack of activity. Feel free to continue commenting and the issue can be reopend on demand.

closed 2017-12-07 by NikolausDemmel