JakobEngel/dso #88: gettimeofday not declared?

fszeliga, 2017-08-24:
Hello,

so I was compiling DSO on a new Ubuntu 16.04.3 install, and got an error: "gettimeofday was not declared in this scope".  Including <sys/time.h> solved the problem, but is this a normal solution? Did anyone else have this problem?

NikolausDemmel, 2017-09-05:
I have not noticed this. I think there are a few using it on Xenial. Maybe you don't have one of the optional dependencies that everyone else does have. Can you post the output of the `cmake` command on the first run inside an empty directory (you could run it again in a new (temporary) folder).

Ideal would be some instructions on how to reproduce this e.g. with a docker image.

NikolausDemmel, 2017-09-05:
PS: where do you add the include?

NikolausDemmel, 2017-12-07:
Closing due to lack of activity. Feel free to comment and we can reopen if needed.

closed 2017-12-07 by NikolausDemmel