JakobEngel/dso #91: Segmentation fault (core dumped)

Echo-uon, 2017-08-31:
I used Ubuntu 16.04 to run the DSO program. I followed every step explained in the Github, but got an error which is 'Segmentation fault (core dumped)'. Can anyone please help with it?

The details can be found below:
START PANGOLIN!
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.X11 Error: GLXBadFBConfig
Initialization: keep 21.8% (need 2000, have 9183)!
INITIALIZE FROM INITIALIZER (2047 pts)!
SPARSITY:  MinActDist 2.100000 (need 2000 points, have 2047 points)!
OPTIMIZE 2047 pts, 2047 active res, 0 lin res!
Initial Error       	A(745022.008418)=(AV inf). Num: A(0) + M(0); ab 0.000000 0.000000!
STEPS: A 0.0; B 19.5; R 2.5; T 10.8. 	REJECT 0 (L -1.00, dir nan, ss 1.0): 	A(1351978.915526)=(AV 13.695). Num: A(901) + M(0); ab -0.000012 1.653431!
STEPS: A 0.0; B 6.7; R 1.1; T 1.8. 	ACCEPT 1 (L -1.60, dir 0.75, ss 1.0): 	A(1271986.808542)=(AV 10.330). Num: A(1490) + M(0); ab -0.000031 2.225756!
STEPS: A 0.0; B 1.1; R 0.2; T 0.9. 	ACCEPT 2 (L -2.20, dir 0.84, ss 1.0): 	A(1217003.589613)=(AV 10.132). Num: A(1482) + M(0); ab -0.000034 2.316089!
STEPS: A 0.0; B 0.0; R 0.2; T 0.4. 	ACCEPT 3 (L -2.81, dir 0.60, ss 1.0): 	A(1202317.785873)=(AV 10.111). Num: A(1470) + M(0); ab -0.000035 2.319128!
LOG 26: 10.116 fine. Res: 1470 A, 0 L, 0 M; (0 / 0) forceDrop. a=-0.000035, b=2.319128. Window 26 (2)
Segmentation fault (core dumped)

NikolausDemmel, 2017-09-05:
I suggest compiling in debug mode and trying to figure out where it crashes exactly.

wowdeath, 2017-11-07:
Hello,I have met the same question as you description, have you solved this problem? If you have solved it, can you tell me how to solve it? Thank you very much.

Echo-uon, 2017-11-07:
Hi,

I just used the Ubuntu 14.04 instead. Then the program runs correctly.

NikolausDemmel, 2017-12-07:
Closing due to lack of activity / lack of more detailed information. If you still have segmentation fault issues, please open a new issue and provide as much information as possible.

closed 2017-12-07 by NikolausDemmel