JakobEngel/dso #92: problems with PNG images

FlorianMaile, 2017-09-04:
I can't get the DSO running with exported PNG images from Matlab. Does anybody experience the same problems like me ? I have a dataset in PNG which runs without a problem. After importing it into Matlab and directly (no changes to the image, just load and write to another file) write to an PNG file these images won't work anymore in DSO.

NikolausDemmel, 2017-09-04:
Haven't tried this, but what exactly does not work? Are there any error messages?

I suggest running in Debug mode and checking at which point it fails. DSO uses opencv to read images, but only supports some formats (https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp). Maybe matlab somehow changes the encoding to something DSO cannot handle?

FlorianMaile, 2017-09-04:
DSO isn't failing. It is just not able to calculate plausible points anymore. It initializes and directly reset for another initializing phase. There are no error messages.

NikolausDemmel, 2017-09-04:
Hm... but the visualization of the images is ok?

Maybe checking the used encoding in the above code would be a good start. Maybe some encoding works for visualization, but isn't properly implemented for the actual image processing.

HagegeR, 2017-09-04:
maybe the image is written as an indexed image and opencv only support
Truecolor images.
i suggest you check some of the options in:
https://www.mathworks.com/help/matlab/ref/imwrite.html#btwhrak

FlorianMaile, 2017-09-04:
The visualization looks fine.

Actually I have some problems to get the DSO build in Xcode for the suggested debug purposes.

NikolausDemmel, 2017-09-04:
To check the image encoding you can also use printf debugging.

For visual Debugging of cmake projects I have had good experience with qtcreator. Also on Osx.

FlorianMaile, 2017-09-04:
I'll receive (with the exported MATLAB image and also with the raw dataset):
> readImageBW_16U called on image that is not a 16bit grayscale image. this may segfault.

FlorianMaile, 2017-09-04:
comparing two example images of the PNG header results in following:

[image]

As you can see the png encoding header is the same?

chabaudmorgan, 2017-09-05:
The hexadecimal part you highlighted may be related to editing time, which is different. However, if you look the next bytes, the file data, they are identical.

> It initializes and directly reset for another initializing phase. There are no error messages.

Are you sure you give DSO the same sequence (watch out for file names, reading file order may be: img1, img10, img11, img12, ..., img2, img20, img21, ...) as for the original png dataset?

When you display the gui, does the displayed image sequence (right image at the bottom) look good?

FlorianMaile, 2017-09-05:
@chabaudmorgan thank you for your hint!

I saved the images like: I1.png, I2.png .... I500.png. I adjusted now my MATLAB export and it worked as expected. Image visualization looked good because the frame rate was high enough to see that the order was wrong.

Thank you all for your very fast help!

NikolausDemmel, 2017-09-05:
I'll close the issue, since it looks like it is solved.

closed 2017-09-05 by NikolausDemmel