JakobEngel/dso #93: Question about the function orthogonalize()

sta105, 2017-09-05:
Dear Jakob Engel,

Thank your for providing this fantastic VO system. Now I am trying to figure out the backend part, and the function "void EnergyFunctional::orthogonalize(VecX* b, MatXX* H)" really puzzled me. 

It seems after we get the increment of the camera poses in sliding window, the system will call this function to project the increment onto the nullspace of the pose. But we don't understand the computation of nullspace of pose
```
SE3 w2c_leftEps_P_x0 = (get_worldToCam_evalPT() * EepsP) * get_worldToCam_evalPT().inverse();
SE3 w2c_leftEps_M_x0 = (get_worldToCam_evalPT() * EepsM) * get_worldToCam_evalPT().inverse();
nullspaces_pose.col(i) = (w2c_leftEps_P_x0.log() - w2c_leftEps_M_x0.log())/(2e-3);
```
Could you give any hint about the purpose of this function? or any reference? Thanks alot!

Sicong Tang

tigerk0430, 2020-01-20:
@sta105 
If you still want to find some references,
see the below well-written blog. (if you're not familiar with Chinese, use google translation.)
https://blog.csdn.net/xxxlinttp/article/details/100080080
https://zhuanlan.zhihu.com/p/29177540

DengueTim, 2025-09-06:
Tong Ling's posts are in English. Updated URL: [https://tongling916.github.io/2020/04/24/DSO-Null-Space/](https://tongling916.github.io/2020/04/24/DSO-Null-Space/)