JakobEngel/dso #94: Errors compiling on ARM (raspberry pi 3)

fluentart, 2017-09-10:
Hey I have compile errors on raspberry pi. I had to mount an external usb flash drive as additional swap to actually get some progress during make.

I tried with and without git submodule update --init

This is the error I get with the ARM SSE2NEON code:

`pi@raspberrypi:~/Desktop/dso_arm/dso/build $ make
Scanning dependencies of target dso
[  4%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o
[  8%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemOptimize.cpp.o
[ 12%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemOptPoint.cpp.o
[ 16%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemDebugStuff.cpp.o
[ 20%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemMarginalize.cpp.o
[ 24%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/Residuals.cpp.o
[ 28%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/CoarseTracker.cpp.o
In file included from /home/pi/Desktop/dso_arm/dso/thirdparty/sse2neon/SSE2NEON.h:123:0,
                 from /home/pi/Desktop/dso_arm/dso/src/OptimizationBackend/MatrixAccumulators.h:29,
                 from /home/pi/Desktop/dso_arm/dso/src/FullSystem/CoarseTracker.h:32,
                 from /home/pi/Desktop/dso_arm/dso/src/FullSystem/CoarseTracker.cpp:32:
/usr/lib/gcc/arm-linux-gnueabihf/6/include/arm_neon.h: In function ‘__m128 _mm_set1_ps(float)’:
/usr/lib/gcc/arm-linux-gnueabihf/6/include/arm_neon.h:5798:1: error: inlining failed in call to always_inline ‘float32x4_t vdupq_n_f32(float32_t)’: target specific option mismatch
 vdupq_n_f32 (float32_t __a)
 ^~~~~~~~~~~
In file included from /home/pi/Desktop/dso_arm/dso/src/OptimizationBackend/MatrixAccumulators.h:29:0,
                 from /home/pi/Desktop/dso_arm/dso/src/FullSystem/CoarseTracker.h:32,
                 from /home/pi/Desktop/dso_arm/dso/src/FullSystem/CoarseTracker.cpp:32:
/home/pi/Desktop/dso_arm/dso/thirdparty/sse2neon/SSE2NEON.h:155:4: note: called from here
  (x)
    ^
/home/pi/Desktop/dso_arm/dso/thirdparty/sse2neon/SSE2NEON.h:324:9: note: in expansion of macro ‘vreinterpretq_m128_f32’
  return vreinterpretq_m128_f32(vdupq_n_f32(_w));
         ^~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/dso.dir/build.make:206: recipe for target 'CMakeFiles/dso.dir/src/FullSystem/CoarseTracker.cpp.o' failed
make[2]: *** [CMakeFiles/dso.dir/src/FullSystem/CoarseTracker.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/dso.dir/all' failed
make[1]: *** [CMakeFiles/dso.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

Has anyone managed to get past this error?

fluentart, 2017-09-10:
Oh I found a pull request, going to test it. 
https://github.com/JakobEngel/dso/pull/80

YJCITA, 2017-09-14:
What's the frequency can it run on pi3? (how many images can calculate every second )

chabaudmorgan, 2017-09-14:
Well, this depends on so many variables...

To give you an idea, on 512x512, fast settings, an iteration (non-keyframe) takes 80ms (keyframe iteration around 150ms).

Did the pull request worked for you @fluentart ?

YJCITA, 2017-09-19:
@chabaudmorgan  I see, thank you very much!