JakobEngel/dso #95: get the filename/ID of image that is used for latest keyframe

FlorianMaile, 2017-09-17:
I'm trying to write my custom OutputWrapper that saves me the inverse depth map to an txt file which is linked to the image that it is correlated with. Actually I'm a little confused when I use the following ID:

`FrameHessian->shell-id`

If I plot the depth map into the corresponding image I can see sometimes a little offset and sometimes it matches perfectly. What is the reason for that ?

perfect match:
[image]

offset:
[image]

Thank you in advance.

s3r637, 2017-09-19:
Hm, could have several reason. (But the most likely is that you do something wrong.)
Could you post a cut out of yours `OutputWrapper`!?

Btw. you should use `fh->shell->incoming_id` (ID passed into DSO) instead of `fh->shell->id` (INTERNAL ID, starting at zero).

Cheers

FlorianMaile, 2017-09-19:
`fh->shell->incoming_id` and `fh->shell->id` is the same for my use case because all images start like `image000001.png`.

Here you can find the cut out of the OutputWrapper:

`        virtual void pushDepthImageFloat(MinimalImageF* image, FrameHessian* KF )
        {
            char fname[512];
            sprintf(fname, "bin/results/invD_%06d.txt", KF->shell->id);
            FILE *fid = fopen(fname, "w");
            for (int h=0;h<image->h;h++)
            {
                for (int w=0;w<image->w;w++)
                {
                    const float invD = image->at(w,h);
                    //if (invD > 0)
                    //{
                        fprintf(fid, "%.12f ", invD);
                    //}
                    //else
                    //{
                    //    fprintf(fid, "NaN ");
                    //}
                }
                fprintf(fid, "\n");
            }
            fclose(fid);
        }`

s3r637, 2017-09-19:
In most cases it is the same. However, ...

Your one-liner is hard to read.
I can't recognize how do you associate the inv depths with the right pixel position.

FlorianMaile, 2017-09-19:
Uploaded the file in the updated post before.

s3r637, 2017-09-19:
Maybe `fh->shell->incoming_id` will fix that issue.

FlorianMaile, 2017-09-19:
No that don't fix that issue. But I found that line in `CoarseTracker.cpp`

		if(debugSaveImages)
		{
			char buf[1000];
			snprintf(buf, 1000, "images_out/predicted_%05d_%05d.png", lastRef->shell->id, refFrameID);
			IOWrap::writeImage(buf,&mf);
		}

which refers to a `lastRef` frameHessian object.

s3r637, 2017-09-19:
How do you validate the table values, in my cases the lines from `CoarseTracker.cpp` works fine.

FlorianMaile, 2017-09-19:
If if look at the images that are created of `CoarseTracker.cpp` I can see that they are all plausible. I implement now my depth map output also there. Coming back to you in few minutes.

closed 2017-09-19 by FlorianMaile

mrwtakev, 2018-04-03:
@FlorianMaile Hello,I now need to get the depth map through the DSO, but I don't know how to change which parts of the code to implement. Can you give me some advice? Thank you very much.